Santroller/Santroller #1: RF as input?

jimmyh98, 2020-05-22:
First of all I would like to thank you for this great project that is very cool because you can use the same device in several systems which is great, almost every day I am looking at the new project commits. I would also like to know if you have plans to add RF as an input. I have a small project that uses UnoJoy as a receiver and a xbox 360 Stratocaster guitar, both with an nRF24L01 module. The idea is very basic because I am a beginner, the guitar sends a 10 byte array with the states of the buttons to the Arduino with UnoJoy, and I would like to use it in your project.

sanjay900, 2020-06-03:
If this is still something you want, it might not be too difficult, since the codebase is already divided into two parts as the Uno is essentially two different microcontrollers in one. I'll have to check if i still have some RF modules around, but if I do end up doing this, I'll have to see which rf modules its worth making it compatible with.

sanjay900, 2020-06-03:
The one problem is that this code base isn't really all that beginner friendly due to it being extremely low level, but hopefully i can add it in a way where it programs both sides for you.

sanjay900, 2020-06-03:
At some point I was going to look into just adding Bluetooth support in a few different ways as well.

sanjay900, 2020-06-07:
I have an idea that might make it so that you can make a receiver that lets you talk to multiple devices, like an xbox receiver. Im going to implement that first, as it simplifies a lot of logic and makes it much easier to implement RF.

sanjay900, 2020-06-08:
Do you have discord? I no longer have access to a PS3, and it would be good if i could get you to test some changes for me at some point, as i am in the middle of rewriting a lot of things and i need to make sure that the changes don't break on the PS3. My discord is sanjay900#5549

sanjay900, 2020-12-01:
So I've finally got a chance to look into this, so ill be playing around with it this week.

sanjay900, 2020-12-08:
I have the RF modules working, now i just need to tidy it up and make it so that the RF module arduino can actually be configured.

sanjay900, 2020-12-16:
ive just pushed a RF transmitter for the micro / leo / pro micro. All i need to do is make it work with the config tool. What you can do now though, is configure a guitar over usb, and then flash the rf firmware, which will read that configuration out. Then to make a receiver you can just flash the standard firmware, but check `config.rf.rfInEnabled = true;` in the main. it will take me a bit to get configuring working with the tool, but this should work for you for now.

sanjay900, 2020-12-16:
As a test, i just built myself a wireless gh live controller and it seems to work just fine, however i also think itd be good to add some power saving features too which i will look at when  i can.

jimmyh98, 2020-12-17:
Wow, I was happy when I saw the email notification and this is wonderful. I left a little bit of the ideas that I had aside due to the various things that happened in the last few months but before stopping my ideas for good, I made a pcb to replace the original lespaul ps3 guitar. I also had the idea of doing something about the battery as a meter with ws2812b leds, because I left a pin on the pcb for that. Anyway the ideas were a little aside and time decreased, but as soon as I have free time I will complete the projects I have and test this wonderful firmware. Thank you Sanjay.
[image]
[image]

closed 2020-12-25 by jimmyh98

sanjay900, 2021-02-09:
The newest config tool can now program and configure guitars with RF chips