Santroller/Santroller #4: Seeking clarification on Pi Pico pin usage (I2C, analog)

DragRedSim, 2022-02-07:
Hi there,
Less of a bug, more of a question. If I'm reading the file /src/pico/pins_arduino.h correctly, I believe the SDA and SCL pins in use are pins GPIO26 and GPIO27? If so, would it be possible/worthwhile considering changing them to pins which are not allocated to two of the three analog inputs to the Pico? I understand this would break compatibility with previously-made adapters, so I'm not expecting changes to be made instantly; however, in the current setup, there is only one pin option remaining to potentially wire in an analog tilt sensor, which could become an issue depending on how people wire their devices up.

sanjay900, 2022-02-08:
If im honest at the moment the pico i2c stuff isnt even working at the moment i dont think, so i doubt any adaptors exist out there for it to be an issue. Most of this codebase is going away in the future anyways, so im happy to pick another pin down the line

sanjay900, 2022-02-08:
Also tbh at this point most people are using standard digital tilt sensors anyways

sanjay900, 2022-02-08:
Weirdly enough, the pico build actually broke I2C for the pro micro as well, so at this point i would just say that that build only supports directly wired guitars

DragRedSim, 2022-02-08:
Indeed, that does correlate with my own findings, both with Pico (not working I2C) and Pro Micro (only works on older builds). I haven't got into debugging why that happens for myself, but I would still keep the pin reassignment noted for when the eventual rewrite happens. I'm personally on the MPU tilt side of things (meaning I would be sharing the I2C pins), but wanted to make note of it for others. 

I did note that the Wii tutorial did specifically not mention the Pico, and I do have a working Pro Micro build; but I wanted to experiment with the Pico, which led to this. As such, no hurry on the rewrite on my part :P

sanjay900, 2022-02-11:
im happy to change the I2C pins around, i'll have an actual think about it at some point when i get to that stage.

sanjay900, 2022-03-03:
heh okay it was something stupid that broke i2c on the pro micro by the looks of things, i havent got a clue about the pico though

sanjay900, 2022-03-04:
pins have been changed
#define PIN_WIRE_SDA 18
#define PIN_WIRE_SCL 19

sanjay900, 2022-03-04:
the new build also should fix i2c on the micro, and might fix it on the pico too, ive had it work on my pico sometimes

sanjay900, 2022-03-04:
i do plan on fixing i2c a little better in the new firmware though.

closed 2022-03-04 by sanjay900