Santroller/Santroller #5: Effects Switch Resetting Position on Maximum setting

ThePixelPolygon, 2022-05-19:
I have the effects switch on my Rock Band controller connected to an analog pin on my Pi Pico. However, whenever I try to set the pickup to the "None" setting (furthest position away from the strumbar), the guitar sets the position to the uppermost (Chorus) setting, as shown in the attached video.

https://user-images.githubusercontent.com/28792503/169197964-8670205f-93e0-40a1-baa4-892158fc24aa.mp4

sanjay900, 2022-05-20:
what happens if you calibrate it so that the maximum state is actually a little lower than all the way up?

sanjay900, 2022-05-22:
the only other thing worth trying is swapping the wires you use for vcc and gnd, i have seen this sorta issue be resolved by that in the past.

ThePixelPolygon, 2022-06-05:
Well, I've just wired up another Rock Band guitar (an RB2 this time around) and I seem to be getting the same issue with that. Both my RB1 and RB2 are using a Pi Pico as the microcontroller.

sanjay900, 2022-06-09:
feel free to try out the preview build and see if it does anything

ThePixelPolygon, 2022-07-06:
I've tried again on the latest version, the axis isn't properly centered and switching the pickup switch to the "new" none position causes the axis to flicker rapidly between chorus and none.

sanjay900, 2022-07-09:
fun, ill restore that to how it was then. I wonder if its something with the specific potentiometer they use then

sanjay900, 2023-06-30:
I know its been a while, but have you got a discord by any chance? if you add me (same username) i could do with someone who has one of these hooked up, going to be looking at adding in proper support for it over the various consoles (each console seems to implement it differently...)

sanjay900, 2024-01-06:
Fixed by [new firmware](https://github.com/Santroller/Santroller/releases/tag/v10.0.1)

closed 2024-01-06 by sanjay900