Santroller/Santroller #6: [Request] Support I2C neck connection

mnkhouri, 2022-05-25:
Some guitar necks, like the GH5/RockBand guitar, use a neck connection that has a 4-pin I2C connection, instead of the typical 6 pin (ground + 5 buttons). In order to read the neck buttons for this neck, the arduino must communicate with the neck over I2C. 

I reverse engineered the I2C communication and documented it in a blog post: https://marc.khouri.ca/posts/2022/GH5-neck-comms.html

The code itself is quite simple, but it does require using the hardware I2C on the arduino, and I'm not sure how many necks use this -- it might be particular to just the Guitar Hero 5 guitar.

So it might not be worth it for you to add support for this. At least, I hope you'll find the information interesting :) and please feel free to close the issue if it's not worth it

sanjay900, 2022-05-25:
Yeah someone else gave me the reverse engineered protocol a while back. I didn't really see a point cause it seems like they cant be polled very fast, and that would limit the speed of everything else

mnkhouri, 2022-05-25:
The original guitar polled the neck every 10ms, but I'm successfully polling at 2ms in my code. I didn't try to find the lower bound -- I'll check it out to see if it can go down as low as 0.5ms.

> the newer firmware is that its a lot more hackable as its platform.io based, so even if i don't get around to it you probably could easily.

Cool! I have another small side project I want to wrap up first, but then I'll probably take a look at doing it

sanjay900, 2022-05-25:
I was also told that if you poll it fast, it just starts reploying with duplicate data and doesn't actually poll faster but I have no confirmation of that

sanjay900, 2022-05-28:
> The original guitar polled the neck every 10ms, but I'm successfully polling at 2ms in my code. I didn't try to find the lower bound -- I'll check it out to see if it can go down as low as 0.5ms. [...]

do note that the new firmware is ages away anyways so it probably won't be ready until your done anyways.
let me know how low it can be polled

sanjay900, 2022-05-28:
The other question is, do you actually need to do all those reads and writes? If it works anything like the wii, i wonder if a two-byte read at 0x12 would give us just the button byte and the first tap bar byte?

mnkhouri, 2022-05-28:
> I was also told that if you poll it fast, it just starts reploying with duplicate data and doesn't actually poll faster but I have no confirmation of that

I updated the code to poll faster, but I'm not sure how I would experimentally verify that each button change is actually registered that quick. Any ideas? I used 0.5ms and even 0.1ms, and the neck doesn't lock-up or fail or anything -- all seems to be working at those polling speeds.

> The other question is, do you actually need to do all those reads and writes? If it works anything like the wii, i wonder if a two-byte read at 0x12 would give us just the button byte and the first tap bar byte?

Nice suggestion!! I've updated the code to do exactly as you said, and it looks to be working great (line 331 in the [updated gist](https://gist.github.com/mnkhouri/e6ac28bc48560b31890ddb61cc7f7a87#file-gh5_neck_comm-ino)). So the wii guitar works similarly? That's very interesting.

sanjay900, 2022-05-29:
Easiest way to test the actual poll rate would be to have the pro mini control one of the frets (have the pro mini ground the pin on the neck pcb), and then time how long it takes for it to respond. Tbh though as long as the neck doesn't lock up than im happy to implement it either way.

>Nice suggestion!! I've updated the code to do exactly as you said, and it looks to be working great (line 331 in the [updated gist](https://gist.github.com/mnkhouri/e6ac28bc48560b31890ddb61cc7f7a87#file-gh5_neck_comm-ino)). So the wii guitar works similarly? That's very interesting.

As far as im aware, a lot of I2C devices work this way, where you write an address and then can just read as many bytes as you want from that address. With wii guitars, there's an initialization sequence (that also sets up encryption but you can disable it with the right initialization sequence) and then you just do a read from 0x00 to read the data. Though they lock up if you poll them too fast!

sanjay900, 2022-06-01:
also, what does the neck do if you dont do the initial 0x00 write and read? does it work just fine?

sanjay900, 2022-06-03:
heh, implementing this actually put me over my max flash storage size on the arduino, and that was after converting the switch statements to jump tables

sanjay900, 2022-06-04:
> Oh noo, I hadn't considered the size. I bet there's an algorithmic way to
> decode the touch pads, I'll let you know if I ever find it but I haven't
> spotted it yet...
>
> This was a PS3 gh5 guitar, but I'd guess all the necks are the same
> because I *think* the necks are interchangeable among consoles.
>
> About the max baud rate and skipping the initial handshake: I'll try this
> next week and report back (and maybe also the test you suggested to find
> the true response rate).
>
> Thanks for trying out an implementation!

I sat there and stared at the necks for ages, and while there definitely
appears to be something vaguely binary related there, I could never come up
with something that made perfect sense
It sounds like wt necks are also compatible with each other I now have an
implementation of that too hanging around, I just need to put a gui over it

sanjay900, 2022-06-09:
feel free to try out the preview build and see if it does anything

mnkhouri, 2022-06-10:
I'm out of town this weekend but looking forward to trying this next week!

mnkhouri, 2022-06-30:
Hi @sanjay900 , I haven't had a good chunk of time to test this and am leaving town for a month -- I'll be sure to try it once I'm settled back in thought!

sanjay900, 2022-09-15:
Im tempted to assume this works, as the DJ hero turntables work in the same way and i can confirm those work

closed 2022-09-15 by sanjay900

mnkhouri, 2022-09-19:
Awesome! Thanks for your work on this. I'll be sure to comment when I get back to my guitar

sanjay900, 2022-09-19:
I should be getting a guitar with one of these necks today too, so ill be sure to test this all and get back to you

sanjay900, 2022-09-22:
Can confirm it is all working now with the latest build.

sanjay900, 2023-09-13:
> I've been reading all @mnkhouri blog posts and this conversation.. first of all I want to thank you both for the work you've put on this! It's awesome!
> 
> Now! I'm in a similar but different spot as Marc was: I have a brandless chinese guitar (which is wireless and pretty much unusable in my place because it keeps disconnecting due to the WiFi interference), so I decided to convert it to "wired". I have pretty much all nailed down (Arduino Pro Micro 5v, wires almost all mapped and ready to solder).
> 
> Problem is that the neck uses I2C too, similar to what you've been discussing here. It's using this [Capacitive TouchPad Chip](https://www.datasheetbank.com/EKT2101-Datasheet-PDF-ELAN) to map the touchpad and frets. I'm no electrical/electronics engineer so my electronics knowledge is very limited. I already implemented the base code to read from it, even the initial handshake the datasheet specifies. Again, thank you very much Mark! your blog is what encouraged me to pursue the reverse engineering path.
> 
> Based on the datasheet I have already mapped the frets bits. I still need to map the touchpad, but I just got the code working last night after three long days of analysis and trial and error.
> 
> I know Mark used two microcontrollers (one for the I2C communication and the other for the controller) but I wanted to challenge myself to integrate all in one single microcontroller, so, for that I will need to modify this firmware in order to support this new feature.
> 
> I would love to know what changed since your last reply @sanjay900 ... It's been a year and you probably finished the new firmware (or maybe not! that's fine too) but I'm wondering if your integration is with two microcontrollers too and if you could provide any insights for me.
> 
> I'll attach a couple photos:
> 
> WIP [image] [image]
> 
> Neck debugging: [image] [image]
> 
> Original Box [image]
> 
> Original board+POGO pins for the I2C neck. [image]

New firmware isn't done yet. Got a link to your repo?

knifesk, 2023-09-14:
Just uploaded it! I been working on the buttons mapping.. I think this is almost ready to go and rock the stage 🤣 

https://github.com/knifesk/guitar-hero-neck

sanjay900, 2023-09-14:
Have you got a discord by any chance?  im sanjay900. I could probably implement this maybe.

sanjay900, 2023-09-16:
> Just uploaded it! I been working on the buttons mapping.. I think this is almost ready to go and rock the stage 🤣
> 
> https://github.com/knifesk/guitar-hero-neck

alright ive probably implemented this, but we will need to test it, its in an unreleased firmware.

knifesk, 2023-09-16:
Hi Sanjay, that's awesome!! And yes I do have discord! Currently I'm on a work trip, but I'll be back on wednesday and I'll be able to try it out! I'll add you to my discord as soon as I'm home! Thank you so much!

sanjay900, 2023-09-20:
> > I was also told that if you poll it fast, it just starts reploying with duplicate data and doesn't actually poll faster but I have no confirmation of that
> 
> I updated the code to poll faster, but I'm not sure how I would experimentally verify that each button change is actually registered that quick. Any ideas? I used 0.5ms, 0.1ms, and even 0.01ms, and the neck doesn't lock-up or fail or anything -- all seems to be working at those polling speeds.
> 
> > The other question is, do you actually need to do all those reads and writes? If it works anything like the wii, i wonder if a two-byte read at 0x12 would give us just the button byte and the first tap bar byte?
> 
> Nice suggestion!! I've updated the code to do exactly as you said, and it looks to be working great (line 331 in the [updated gist](https://gist.github.com/mnkhouri/e6ac28bc48560b31890ddb61cc7f7a87#file-gh5_neck_comm-ino)). So the wii guitar works similarly? That's very interesting.

FYI if you ever wondered
[image]

Heres the actual numbers for the poll rate of the gh5 i2c neck frets.