Santroller/Santroller #7: Feature request: Stage Kit

sonik-br, 2022-12-26:
Hi,
There's been any attempt (or interest) at simulating a Rock Band Stage Kit?

From what I've found, this looks to be it's usb vid and pid `0x0e6f` `0x0103`

https://github.com/jummama/wxStagekit/blob/master/stagekit.c#L132

sanjay900, 2022-12-26:
quite a few things need to be implemented before i can attempt to look at this

sanjay900, 2022-12-27:
> Ok, thanks for the reply.
>
> I've just got an x360 console that can use UsbdSecPatch.
> I will take a look if at least the game detects a pico flashed with the
> correct vid and pid.
>
> I will update here with my findings. But feel free to close this issue if
> needed.

There's some new led code on the horizon that will be extremely similar to
how the stage kit works so I should be able to add it then

sanjay900, 2023-01-08:
i do suspect ill have to find someone with a stage kit though because xbox 360 games don't usually use vid and pids for detecting devices, its done via the capabilities which i dont have a copy of

sonik-br, 2023-01-08:
I've not ever seen one before. It's super hard to find.
I only know that the game sends commands to it via "rumble"

And someone used GIMX to trick the game into thinking an arduino is an Stage Kit. But no source code...
https://www.reddit.com/r/Rockband/comments/jt2jfv/comment/gc62mt4/

sanjay900, 2023-01-09:
[image]
got it.

sanjay900, 2023-01-09:
https://github.com/sanjay900/Ardwiino/commit/092932778dcec0370992d0ced6a1dbd46a3b3202
https://github.com/sanjay900/Ardwiino/commit/55e0829da1000ddd00812a0ddf368372604dba4d

closed 2023-01-09 by sanjay900

sanjay900, 2023-01-09:
The plan will be for the stage kit to work the same way as other leds in the new firmware, where you can bind an event from the stage kit to one or many leds on an apa102 strip, or to a output pin on your micro controller for say the fog machine. I already support things like overlapping LEDs and stuff in the firmware as well, so youd be able to do something like binding the 8 different colours to 8 leds and that kind of thing.

sonik-br, 2023-01-09:
This is absolutely amazing!
Thank you very much for implementing this. Been looking for a way to recreate the stage kit for a long time.

I'm probably going to drive make it drive my "ambilight" first. (already have a led strip behind my tv, controlled by arduino)
And ultimately recreate the same experience from the stage kit with a white strobe and lights firing to the ceiling.
Unfortunately the xbox console I've got recently was in a bad shape and died. Now I need to find another :(

sanjay900, 2023-01-09:
what LEDs are you using on your ambilight?

sonik-br, 2023-01-10:
I'm using an adalight based firmware with WS2812B leds

sanjay900, 2023-01-10:
that does get a bit unfortunate since i dont support WS2812Bs currently, due to some issues with them and usb, though i may eventually add support for  em to the picos

sonik-br, 2023-01-10:
No problem at all. This goes out of scope.
Just having the device talking to the game and having a way to get the stagekit commands will help A LOT!
I might interface with it externally in some way.

sanjay900, 2023-01-10:
I was talking to one guy that wants to try and hook it up to DMX lighting haha, that will be a fun project to watch

sonik-br, 2023-03-28:
That would be amazing!
Having the stage kit info on digital pins would open a lot of opportunities to use this data.
Thanks for doing this.

Btw, any news on the DMX project?

sanjay900, 2023-03-28:
Not yet, thats sorta on my list of things to look into after the new firmware is out haha