Santroller/Santroller #8: rpi pico won't upload

exur00, 2023-03-21:
I am using a pico, and when I put it in bootsel before opening guitar-configurator I can select the pico, but it then instantly gets stuck on "Programming Status: Waiting for Reconnection (0.00%)". In the terminal it also prints "Error retrieving serial".
If I ignore the recommendation I don't put the pico in bootsel, it sees a generic serial device, but it only allows me to choose arduino models, not the pico, so that's also not the way to go.
I've also tried 2 other picos and got the exact same error.
Any idea what could cause (and fix?) this? If there's any extra information I could give you that could be of use, ask away.

sanjay900, 2023-03-21:
What OS? If your on Linux, then there are extra instructions to follow, they are linked in the guide but maybe I need to make them more obvious
https://github.com/sanjay900/guitar-configurator/issues/32

closed 2023-03-21 by exur00

sanjay900, 2023-03-21:
> I am using linux, and that did indeed solved my issues, now it works great!
> I didn't notice the link in the guide, so making it more obvious is maybe
> a good idea.
> Thanks a lot for the quick response!

I intend to automate this process when the new config tool comes around,
but maybe I'll add a link to this within the tool itself or something for
Linux

sanjay900, 2023-04-11:
Just added some more notes here:
https://sanjay900.github.io/guitar-configurator/guides/linux.html
if you plan on using ps3 mode and stuff, might want to follow the updated instructions in here.