Santroller/Santroller #10: Names

kizzoknight, 2023-05-18:
Not sure if this is the best place to ask or if you will reveal you secrets but I'm working on a pi Pico project and was wondering how you changed the device name if you could help at all that would be great no worries if not

sanjay900, 2023-05-18:
I do it  by modifying the device descriptor, but then i give my self direct access to the USB layer to do that, so i wouldn't know how to do that with the standard arduino ide or platformio setup that your probably using. I *think* arduino lets you do it by modifying boards.txt or something, but i have no idea youd have to read up on the documentation for whatever platform your using.

kizzoknight, 2023-05-18:
I'm currently using device descpitor to define how many axis and buttons to show up on windows so could I add it to that?

sanjay900, 2023-05-18:
different descriptor that, thats the HID descriptor, its the USB device descriptor you need to modify.

kizzoknight, 2023-05-18:
Okay cheers might have a look into it would be cool to get working but it's only a little project for myself so if I can't figure it out it's not that bad

sanjay900, 2023-05-18:
If you have your project on github i can give you some pointers?

kizzoknight, 2023-05-18:
It's currently just a few files on my pc but I might put it on GitHub
tomorrow. Cheers

sanjay900, 2023-05-18:
just using the arduino ide, or platform.io?

kizzoknight, 2023-05-18:
Currently just Arduino IDE and editing a library for Pico game pads

sanjay900, 2023-05-18:
got a link to the lib?

kizzoknight, 2023-05-18:
https://gitlab.com/realrobots/PicoGamepad

I'm still pretty new to code this in depth so thought this was a good way to learn. I've done some low level code before

sanjay900, 2023-05-18:
from what i understand, you should be able to just modify the name in `boards.txt` and that should then flow through to your device

sanjay900, 2023-05-18:
I think  for you (assuming windows) that file will be in `%localappdata%\Arduino15\packages\arduino\hardware\mbed_rp2040\3.0.1\boards.txt`  if you are using the mbed board
If you are using `arduino-pico` then it should be in `%localappdata%\Arduino15\packages\rp2040\hardware\rp2040\3.2.0\boards.txt` 

If you are on linux or macos, then it would instead be in `~/.arduino15/packages...`

sanjay900, 2023-05-21:
Im going to close this issue since it isnt really related to the project, but if you still need a hand feel free to reach out on discord or something, but hopefully thats all you need to get it working

closed 2023-05-21 by sanjay900

kizzoknight, 2023-05-21:
I apologize for re opening this issue but what is your discord name. The boards.txt only works for regular Arduino boards

sanjay900, 2023-05-21:
sanjay900#0001