Santroller/Santroller #76: [Help]: Whammy Bar not sending midi

ncint, 2023-09-10:
### Please describe your issue.

Hello! 

I configured everything ( p2 guitar hero) and the whammy bar is not sending **midi**. Everything else is working.

I calibrated the whammy in the guitar configurator and it reads the input so i guess it´s not the potentiometer...
I tried  note send and Control message and no luck.  Any thoughts?

- By the way , Is there any way to send CC messages? (ex CC23)

Thanks for you work!!

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2024-01-06:
Sadly MIDI support didn't make the cut for the new firmware yet, its something that will be worked on next most likley

sanjay900, 2025-11-30:
I don't see this being something i work on any time soon, theres just too much other stuff to work on before i can figure out midi stuff some day

closed 2025-11-30 by sanjay900