Santroller/Santroller #100: [Issue]: Download link not working

iMrDJAi, 2023-09-19:
The download link (https://github.com/sanjay900/SantrollerConfigurator/releases/latest) on the documentations page (https://github.com/sanjay900/SantrollerConfigurator/releases/latest) leads to a non-existing repo.

sanjay900, 2023-09-20:
which documentations page are you referring to here

sanjay900, 2023-09-20:
Ah, your looking at the unreleased documentation. That code is not out yet, please use https://sanjay900.github.io/guitar-configurator/

closed 2023-09-20 by sanjay900

sanjay900, 2023-09-20:
I have updated the links to make it clear that this firmware is not released yet

iMrDJAi, 2023-09-20:
@sanjay900 Sorry, I've been referring to https://santroller.tangentmc.net/#download-link. Any ETA for the new firmware tho? I'm interested in using my DualShock 4 controller on my Xbox 360, is that supported by the Guitar Configurator?

sanjay900, 2023-09-20:
No ETA, I can't actually remember if I supported inputs from PS4 controllers

iMrDJAi, 2023-09-20:
@sanjay900 I saw an image from https://github.com/Sk1ller1993/J-Runner-with-Extras demonstrating a PS4 controller working on xbox 360, I assume this is Guitar Configurator setup right?

sanjay900, 2023-09-20:
nope, that would be something totally different.

sanjay900, 2023-09-20:
Okay i do have support for PS4 controller inputs it seems though

iMrDJAi, 2023-09-20:
@sanjay900 Alright, I have a PI Pico, I'll experiment with it and let you know if it worked out.

sanjay900, 2025-06-26:
What exactly do you mean, what can't you find

iMrDJAi, 2025-06-26:
@sanjay900 PS4 input not shown on list:

[image]

(Or perhaps it's labeled differently?)

sanjay900, 2025-06-26:
USB Host input