Santroller/Santroller #104: PS2 5V Wireless Dongle Issues

rgstoian, 2023-11-23:
I recently bought a PS2 World Tour guitar. The dongle is rated 5V 35mA, which is a bit weird seeing that most documentation refers to 3.3V being used.

[image]

I wired up the dongle using your guide, a 5V Arduino Uno and a couple of these TTL logic converters: [image] 
I only have 4 bidirectional channels, so I put Command/MOSI on a unidirectional 5V->3.3V pin. In this configuration, the dongle lights up and pairs to the guitar (and the onboard dongle LED reacts to button presses on the guitar), but shows up as not connected and none of the buttons work.

If I skip the logic converters and wire it directly, keeping the 1k resistors on Data/MISO and Ack, it works fine.

Furthermore, I'm seeing that the cheap PS2 to USB converter I have (which works fine with Clone Hero, but not with GH3 because it's detected as a DirectInput device) already outputs 5V to the power pin of the controller socket.

Are my logic converters bad or will this dongle only work on 5V? I want to get a Pico to make a more compact adapter, and I'm not sure if it will work properly on 3.3V.

sanjay900, 2023-11-23:
Weird! What happens if you use the logic converter but hook only power up to 5v instead of 3.3v? If that works then the Pico should be fine.

sanjay900, 2023-11-23:
In saying that, if you don't intend to use this on a ps2, I would highly recommend direct wiring, the latency on the wireless PlayStation instruments is pretty bad

rgstoian, 2023-11-23:
> Weird! What happens if you use the logic converter but hook only power up to 5v instead of 3.3v? If that works then the Pico should be fine.

Still not seen as connected, but pairing and receiving inputs on the status LED. I'll just order a Pico and a better, fully bidirectional logic converter and report back when I'll get them wired up on both 3.3V and 5V. I guess Pico clones (I hate that the official one has micro-USB) are fine too as long as they have identical specs?

> In saying that, if you don't intend to use this on a ps2, I would highly recommend direct wiring, the latency on the wireless PlayStation instruments is pretty bad

Honestly it feels fine after calibrating, I'm playing casually anyway so wireless convenience wins out in my case.

sanjay900, 2023-11-23:
It would be interesting to see if it just works with a Pico anyways, maybe something is just wrong with your ttl adapter

rgstoian, 2023-12-02:
I got my Pico and everything is working fine when connected on 3.3V, even if the dongle's specs claimed otherwise. Turns out my logic converters were duds. 

Thanks for the support, can't wait for the new version!

closed 2023-12-02 by rgstoian

RaoulBe, 2024-03-21:
Hi. 

I tried to build a PS2 adapter following the guide, and I tried it out with 2 PS2 guitar dongles (GHWT and Kramer Striker, both specified at 3.3V). Both leds shine faintly, and guitar will not connect. (issue #153). Any thoughts on what I might have done wrong? I described my config in #153.
It could be something with current or voltage. How did you build the working one on Pico? Did you follow the same guide as I did? Did you keep the 1kOhm resistors?

Thanks

/Raoul

sanjay900, 2024-03-24:
I guess that's funny. The adapter that states 5v works fine on 3.3v but the adapters that specify 3.3v only work at 5v lol