Santroller/Santroller #114: [Issue]: Santroller is unable to locate `libusb-1.0.so` on Gentoo Linux

JamesIsWack, 2024-01-07:
### Please describe your issue.

Upon launching Santroller, a message appears saying that `libusb-1.0.so` cannot be found, even if it is installed. Not sure about other distros, but the package manager Portage in Gentoo installs the `.so` output files into `/usr/lib64`, which may be causing this problem. A fix would be to statically link the binary, however this would increase the size. Stripping the resulting executable would take some away from that. 
[image]
[image]

### Version

yes

### What operating system are you using?

Linux

### os-other

Gentoo Linux

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2024-01-07:
I'm using https://github.com/LibUsbDotNet/LibUsbDotNet to handle all this, so i'm not really sure how easy that will be to fix.

JamesIsWack, 2024-01-07:
I did suggest statically linking the binary, which wouldn't cause this to happen.

sanjay900, 2024-01-07:
Yes but I don't know if I can easily do that with dotnet

JamesIsWack, 2024-01-08:
You could try using [ILMerge](https://github.com/dotnet/ILMerge)

sanjay900, 2024-01-08:
That's not what ilmerge is for I don't think. I could totally just include
copies of the libusb libraries in the executable like I do for other libs
but there's gotta be a reason libusb doesn't just get picked up

JamesIsWack, 2024-01-08:
Its not being seen because its looking in `/lib64`, and because im on
Gentoo. libs are installed with the `/usr` prefix. so instead of `/lib64/
libusb-1.0.so` its `/usr/lib64/libusb-1.0.so`.

sanjay900, 2024-01-08:
That sounds like a bug report for libusbdotnet, though I may just patch it
myself at some point

closed 2024-02-12 by sanjay900