Santroller/Santroller #116: [Issue]: Up-strum Glitching

Banzah, 2024-01-08:
### Please describe your issue.

First of all, thank you so much for your hard work with this project. It's allowed me to mod my first guitar heri guitar with great results.

My only issue is that my up-strum is occasionally glitching out and lagging and sending a double strum. Any idea why this would be? Down strung is completely fine.

Thanks in advance.

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-01-08:
That's odd, is this happening on console or just on windows, and xinput or direct

Banzah, 2024-01-08:
It's happening on every input; I'm pretty confident it's the switch after some testing. I've ordered a new switch, and it'll be arriving in a few days. 
Will swap out the switch and hopefully that'll work.

sanjay900, 2024-01-08:
Debounce should help with a bouncy switch if you want a fix before it arrives, it would just cancel out the second input

closed 2024-01-08 by sanjay900