Santroller/Santroller #117: [Issue]:Cant download the guitar configurator

ej9904, 2024-01-08:
### Please describe your issue.

goes to download but says file isnt trusted, havent been able to find a way around it.

### Version

yes

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-01-08:
Thats going to be something with your browser, it hasn't been downloaded enough times to make it to the trusted software database but you can always override it.

closed 2024-01-08 by sanjay900

ej9904, 2024-01-08:
Hi, sorry if this is intrusive or bothersome but can you tell me how i could overwind it Sent from my iPhone

sanjay900, 2024-01-09:
what browser are you using?

ej9904, 2024-01-09:
I figured it out after I emailed you, thank you for responding though, i appreciate the custom service and support, thank youSent from my iPhone

sanjay900, 2024-01-09:
fantastic news

ej9904, 2024-01-12:
How do i hook the guitar up to ps4 rock bandSent from my iPhone

ej9904, 2024-01-12:
> Set the guitar to rock band mode, it should just work on a PS4.

Ran into new problems, the guitar can connect now, but the option button wont work, strum wont work.Sent from my iPhone

sanjay900, 2024-01-12:
strum not working is weird, and i presume the options button is the start button so its very odd if that doesnt work. Is this an adapter or a directly wired guitar?

ej9904, 2024-01-12:
Directly wired controller, i got the solderless kitSent from my iPhone

sanjay900, 2024-01-12:
best hitting up the person you brought the kit from, I don't really provide support for kits

ej9904, 2024-01-12:
Okay thank you for tryingSent from my iPhone