Santroller/Santroller #118: [Issue]: Initial configuration stuck at 90% "waiting for device"

Jedievvie, 2024-01-10:
### Please describe your issue.

initial configuration will not finish using pi pico, stuck at 90%.

### Version

yes

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-01-10:
did you only recently download the tool, or did you download it a while ago but only raise an issue now?

sanjay900, 2024-01-10:
As for a fix
1. unplug the guitar / adapter
2. run https://www.nirsoft.net/utils/usb_devices_view.html as admin
3. uninstall everything with vid 1209 and pid 2882
4. plug it back in

Jedievvie, 2024-01-10:
Just installed for the first time today and the issue started immediately.

sanjay900, 2024-01-10:
I wonder what's caused that bug to happen

Jedievvie, 2024-01-10:
Maybe important to mention but I’m using a pico modding kit by retrocult to mod my guitar for clone hero, could the hardware I received be the issue if there’s something faulty about it? I’ve also had issues getting my computer to even recognize the device without saying it’s malfunctioned giving me code 43 in the device manager that’s why I ask.

sanjay900, 2024-01-10:
Shouldn't be that, though also make sure you aren't using a USB hub

sanjay900, 2024-01-10:
Let me know if the usbdeview fix works for you though!

Jedievvie, 2024-01-10:
So i may have made a mistake. On the tutorial video for the installation
there was a link for santroller and one for guitar configurator, it says to
use guitar configurator. USBDeview helped with the pc recognizing something
is plugged in but the raspberry pi no longer shows up in my file explorer
and in USBDeview it shows as santroller. Santroller recognizes the
raspbeerry pi but to be honest i have no idea how to configure the guitar
using it as its nothing like the software i was supposed to be using and
i'm not the brightest when it comes to programming things[image: image.png]

Jedievvie, 2024-01-10:
> i'm not the brightest when it comes to programming things[image:
> image.png]

Ive got everything working now!

sanjay900, 2024-01-10:
fantastic, sorry about it not just working!

closed 2024-01-10 by sanjay900

sanjay900, 2024-01-13:
Fyi since I feel like the whole usbdeview thing was super complicated, Ive made a single button that does all that logic now hopefully!