Santroller/Santroller #120: [Issue]:  Santroller flat out crashes when I press configure on the menu screen.

JMartinez9820, 2024-01-10:
### Please describe your issue.

For some reason, Santroller just simply crashes when I press configure. Guitar Configurator doesn't do this, and what's weird is that it was working earlier... did I do something wrong or did something from Windows Update break it? I hope it's not because I tried to do a guessing game with the D+ pin for the USB host input option for my RCM v3 adapter (since I haven't gotten a response back yet)... I'll likely post a video if necessary

### Version

v10.0.27

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2024-01-10:
That's exactly what it is - though I'll have to work out why that makes it crash.

closed 2024-01-10 by sanjay900

JMartinez9820, 2024-01-12:
I have now resolved my issue, just needed to delete the santroller stuff in the roaming folder. Thanks those who helped me on discord.