Santroller/Santroller #121: [Issue]: Problems with PS2 controllers

Tohsaka300, 2024-01-10:
### Please describe your issue.

I have 2 PS2 controllers, one official type A and one generic.

The official is recognized as a guitar. And with both doesn't work the L1 and R1 buttons.
In Xinput, the L2 and R2 are mapped to the LB and RB buttons instead of the L1 and R1, and some weird signal from the left stick, trying to press the d-pad in some strange directions.

I returned to V9.4.9 and it work.

### Version

v10.0.27

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

PS2

sanjay900, 2024-01-10:
Guitars would do that yeah, weird though

sanjay900, 2024-01-11:
Out of curiousity, were you holding dpad left when you tested either of these controllers? Thats the only thing that  would tell it its a guitar and not a gamepad.

sanjay900, 2024-01-11:
alright with v10.0.30, the controller seems to work for me
you will need to remove and re-add the ps2 controller inputs thing in order for it to pick up the fixed bindings though when the release comes out

sanjay900, 2024-01-11:
Actually thats very odd, the guitar code should only ever try to trigger for dualshock one controllers, and only then will it look at the dpad

sanjay900, 2024-01-11:
a v10.0.32 is building with a few misc PS2 fixes, give that a go when it comes out

sanjay900, 2024-01-11:
Try the latest version out

sanjay900, 2024-01-11:
Sounds like it's specific to the type a. I have one around but its broken, I'll try to fix it and then work out what's going on there.

Tohsaka300, 2024-01-11:
I tried the v10.0.41, it recognized very great, one minimal detail is the pressure sensitive buttons, it is recongized as a DS1. I think that's the fact of the analog mode is togglable. (Analog LED turn on or off if You press the Analog button).

And even a Blueretro adapter recognizes as a DS2, but with the triggers of any BT controller are treated as digital buttons.

All of that doesn't matter, but the fact is, the controller work as usual.

Thank You so much.

sanjay900, 2024-01-11:
It *should* turn on analog mode on startup and detect it though, I'm gonna repair my type a controller and figure out why it's detecting that one as a ds1 instead of a ds2.

I'd expect the triggers to show up if you scroll further down - they are grouped under pressure sensitive buttons

Tohsaka300, 2024-01-12:
The triggers thing is due to the Blueretro firmware, but is an optional detail.

sanjay900, 2024-01-12:
Ah okay thats all good then

Ill let you know when a fixed DS2 firmware is out

sanjay900, 2024-01-12:
can you please test v10.0.43 when it comes out?

sanjay900, 2024-01-13:
.43 or later should make the PS2 controller show as a ps2 controller

Tohsaka300, 2024-01-13:
Is shows as a DS2, but the L2 R2 buttons are not recognized, only in the configurator with the addition of the pressure levels in those and the rest of the pressure sensitive buttons.

sanjay900, 2024-01-13:
As long as the detection bit works I can fix the rest

sanjay900, 2024-01-13:
what aren't L2 and R2 recognized in exactly?

Tohsaka300, 2024-01-13:
None of those above. In other gamepad testers, the L2 R2 buttons have no action, no digital press and no axis movement.

sanjay900, 2024-01-14:
Yeah its just started happening to me now out of nowhere

sanjay900, 2024-01-14:
i think i understand why

sanjay900, 2024-01-14:
Yeah okay i see the problem will be fixed in next release

sanjay900, 2024-01-14:
give that one a go, should all be fixed now

Tohsaka300, 2024-01-14:
Yeah, it works. Even with the Blueretro adapter.

In DS2 recognizes the pressure in L2 R2 flawlessly.

Thanks

closed 2024-01-14 by Tohsaka300

sanjay900, 2024-02-18:
Would you be able to try the latest version of santroller for me? I made some more PS2 controller changes to make things more correct but just want to confirm I didn't break this. Works with both my ds2 controllers but just want to be sure.

Tohsaka300, 2024-02-19:
No problems detected. Tested in v10.0.130. Pressure triggers works as usual.