Santroller/Santroller #122: [Issue]: Adapter Stops Being Detected

lunieetoons, 2024-01-11:
### Please describe your issue.

Santroller picks up my adapter when I plug it in but when I hit configure, my adapter stops being detected. Anyone know a fix for this? I’m using a RetroCultMods wii adapter.

### Version

10.0.27

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-01-11:
Try 10.0.29

lunieetoons, 2024-01-11:
When I load Santroller, it comes up under the devices as Ardwiino Raspberry PI Pico like normal but after I hit configure, I hear that disconnect sound even though it’s still plugged in and it disappears from my devices and printers list. It comes up as a keyboard in my devices and printers list (not sure if that’s normal.) I’ve tried just closing Santroller and trying again but it does the same thing everytime.

lunieetoons, 2024-01-11:
Also, when it comes up under devices and printers it says that the Ardwiino drivers need to be reinstalled?

sanjay900, 2024-01-11:
thats normal

sanjay900, 2024-01-11:
go into device manager, set it to sort by container and send a screenshot

lunieetoons, 2024-01-11:
[image]

sanjay900, 2024-01-11:
weird thats 100% the old firmware, so its not even trying to write the new firmware

sanjay900, 2024-01-11:
is the old tool open in the background still?

lunieetoons, 2024-01-11:
Correct. It'll say writing but then it disconnects and just stays at 0%. Before I hit configure, it shows up as a USB flash drive but it's gone right after I hit it and I hear the sound. This is how it's coming up now without me trying to configure. Santroller says Ardwiino is running version 9.3.1
[image]

lunieetoons, 2024-01-11:
This is how device manager looks after I hit configure. Ardwiino's completely gone even though it's still plugged in. 
[image]

sanjay900, 2024-01-11:
okay what i would try is unplug it, hold the little button and plug it back in, it should then show up as the usb drive and you can try programming it in that mode
It will wipe your config however - you will want to pick wii adapter mode
and use pins GP18 and GP19 for the wii adapter

lunieetoons, 2024-01-11:
I held the button and then tried again with Santroller multiple times but there's been no change. It stops detecting it as Ardwiino and then RP2 Boot comes up under my devices but the status stays at 0%.

sanjay900, 2024-01-11:
If you hold the button when plugging it in, it should never show up as an ardwiino in the first place.

Im talking about the one inside the little hole in your adapter

sanjay900, 2024-01-11:
Anyways if you need more help, hit up retrocult, I don't really intend to support products as its difficult to do so