Santroller/Santroller #123: [Issue]: Mac OS Application Won't Start

MarbleMuscle, 2024-01-18:
### Please describe your issue.

Mac OS claims the application is "broken" every time I try to open it.

### Version

Latest (also tried the last few

### What operating system are you using?

Mac OS

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2024-01-18:
macOS detects the application as damaged as apple requires a developer license to properly sign apps
you can run xattr -r -d com.apple.quarantine /Applications/SantrollerConfigurator.app from a terminal to fix this

MarbleMuscle, 2024-01-18:
sorry i don't mean to be annoying but how do I do that?

sanjay900, 2024-01-18:
Open the `terminal` application, and then paste that in

MarbleMuscle, 2024-01-18:
[image]

MarbleMuscle, 2024-01-18:
Not enough arguments for option -d. Expected at least 2 but got 1

usage: xattr [-l] [-r] [-s] [-v] [-x] file [file ...]
       xattr -p [-l] [-r] [-s] [-v] [-x] attr_name file [file ...]
       xattr -w [-r] [-s] [-x] attr_name attr_value file [file ...]
       xattr -d [-r] [-s] attr_name file [file ...]
       xattr -c [-r] [-s] file [file ...]

The first form lists the names of all xattrs on the given file(s).
The second form (-p) prints the value of the xattr attr_name.
The third form (-w) sets the value of the xattr attr_name to the string attr_value.
The fourth form (-d) deletes the xattr attr_name.
The fifth form (-c) deletes (clears) all xattrs.

options:
  -h: print this help
  -l: print long format (attr_name: attr_value and hex output has offsets and
      ascii representation)
  -r: act recursively
  -s: act on the symbolic link itself rather than what the link points to
  -v: also print filename (automatic with -r and with multiple files)
  -x: attr_value is represented as a hex string for input and output

sanjay900, 2024-01-18:
Where is SantrollerConfigurator.app located at the moment?

MarbleMuscle, 2024-01-18:
I thought it was in my applications, but now I think its just in my downloads

sanjay900, 2024-01-18:
youd have to revise that command for wherever the executable is

MarbleMuscle, 2024-01-18:
YOOO i moved it to my applications, ran the original command again and it worked!!

MarbleMuscle, 2024-01-18:
thank you so much!!

sanjay900, 2024-01-18:
no problem! Ill see if i can work out why this is necessary though with a macos vm later

closed 2024-01-18 by sanjay900