Santroller/Santroller #124: [Issue]: RPCS3 Automatically Disconnects a Santroller Guitar (Auto Reboot Failing?)

Gocnak, 2024-01-18:
### Please describe your issue.

Hello!

First of all, thank you so much for writing this amazing software. The configurator works perfectly with my pi pico-based mod kit I bought, and it's very intuitive to use. Your work is incredible!!

I seem to be hitting a problem that's exclusively in the Santroller firmware when using it in RPCS3. Upon booting into Rock Band 3 on RPCS3, [your code in the emulator](https://github.com/RPCS3/rpcs3/blame/master/rpcs3/Emu/Cell/lv2/sys_usbd.cpp#L250) properly detects the controller and sends the 0x03f2 control transfer message to the Santroller, prompting the controller to reboot.

However, there appears to be an issue with the rebooting. While the controller properly disconnects, it seems to fail to *reconnect* to RPCS3 in the proper mode. Stopping the game and restarting it prints out `sys_usbd: Found device: Guitar Hero Guitar`, but none of the input profiles (Dualshock 3, Dualshock 4, DualSense, XInput, MMJoystick, SDL) inside of RPCS3 seem to detect it, and thus no inputs actually go into the game. Is there a specific thing to do in the Santroller configurator or RPCS3 to have it show up after this swap?

Physically unplugging the controller while the game is running and plugging it back in makes the controller work again, but under the configured XInput profile I configured my controller to use in the first place. This is probably due to the Santroller input detector only running in the constructor of the `usb_handler_thread()`.

As unplugging and replugging the device is not ideal, and restarting an emulator/emulated game just to get into the right mode is also not ideal, could it also be possible to add a setting in the configurator to disable this RPCS3-specific `Found device: Santroller` consoleType swap? It makes sense to do for a PS3 or other actual hardware console, but from my testing, the XInput handler seems to work flawlessly in RPCS3 when I manually replug the controller in, and was not a hassle to set up bindings for in the emulator as well.

### Version

v10.0.68

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-01-18:
The input profiles aren't supposed to detect it. Remove all your input profiles, you don't need them and they will be interfering

Gocnak, 2024-01-18:
Thank you for the information!

To be clear, the Santroller is still not sending inputs to the game on first boot (when it reboots into the right mode) *nor on the second game boot* (when it's detected as a guitar hero guitar). Is there perhaps a setting in RPCS3 that would be needed to have it pass through to the controller?

Hotplugging will be amazing to get as well, I'm looking forward to hearing about the progress there. Thank you again.

sanjay900, 2024-01-18:
I'm not quite sure exactly what the setting is, but I do know that the default settings work. I would try disabling all your mappings and see if that works, i have not quite yet worked out what setting it is that stops it from working.

sanjay900, 2024-01-21:
Interesting
it seems guitar hero guitars dont work but rock band ones do
so i must be emulating something slightly wrong specifically for guitar hero guitars. Since your playing RB, setting it to rb mode should just make it work

sanjay900, 2024-01-21:
I am just going to disable that mode for anything that isnt a turntable i think.

sanjay900, 2024-01-21:
Fixed with 10.0.73 when that comes out

sanjay900, 2024-01-21:
I  do still plan on RPCS3 hotplug for turntables but i have to remember how c++ works a bit more lol

closed 2024-01-21 by sanjay900

sanjay900, 2024-01-22:
https://github.com/RPCS3/rpcs3/issues/15073
If you were curious, turns out the games just use totally different APIs only for talking to guitar hero instruments, hence why they specifically don't support passthrough

Gocnak, 2024-01-22:
Very interesting! One workaround that a friend and I did was to just manually install the WinUSB drivers onto the device and RPCS3 was detecting it then, but glad to see it was a simple mode switch to RB. Thanks for following through on all of this, you rock!