Santroller/Santroller #126: [Issue]: impossible to bind the d-pad (strum) of a PS3 guitar through USB host

Proofeek, 2024-01-23:
### Please describe your issue.

I am trying to connect a PS3 guitar with a dongle to an Xbox 360 using Santroller and USB Host. All buttons binds perfectly except for the d-pad and whammy. Apparently, this is related to the fact that they are axises. When I connect the guitar directly to the PC, the gamepad tester shows that the d-pad corresponds to different values of axis 9, and the whammy corresponds to axis 2. 
[image]

By the way, it is not possible to bind the slider via USB Host Input at all (axis 5). Also, I'm not sure, but it seems to me that the tilt is not working quite correctly.

I tried to do the same with drums, and everything binds except for the d-pad (for the same reason).

If I wrote something unclear, I am ready to provide additional information and answer any questions.

### Version

v10.0.78

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-01-23:
They aren't actually axis, that's just how your gamepad tester shows them

sanjay900, 2024-01-23:
when it releases v10.0.79 will fix the dpad

Proofeek, 2024-01-24:
d-pad still doesn't work in v10.0.79

How to calibrate whammy and tilt? As far as I understand, there is no calibration through USB Host Input.
[image]

sanjay900, 2024-01-24:
You don't need any of these binds, remove them

sanjay900, 2024-01-24:
instead, just add the "Usb host inputs" setting.
[image]

sanjay900, 2024-01-24:
not sure why binding it the way you did resulted in no calibration, ill have to look into that but this way does

Proofeek, 2024-01-24:
Now everything works. After saving the settings the d-pad started working too! Thank you!
Only slider did not bind.

sanjay900, 2024-01-24:
Sliders are weird, I'm not totally shocked that doesn't work, I'll have a look in a bit, probably just missing the math to properly translate it.

Does the slider react in the tool at all?

Proofeek, 2024-01-24:
No

sanjay900, 2024-01-24:
this a WT or a GH5 guitar?

Proofeek, 2024-01-24:
WT

sanjay900, 2024-01-24:
Ah yeah, probably not handling that properly

sanjay900, 2024-01-24:
Can you send me a copy of `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`

Proofeek, 2024-01-24:
[config_data.txt](https://github.com/Santroller/Santroller/files/14033285/config_data.txt)

sanjay900, 2024-01-24:
can you hit the restore to arduino button, and then extract this and drag the .uf2 file onto the usb drive that shows up?
[firmware.zip](https://github.com/Santroller/Santroller/files/14033365/firmware.zip)

Proofeek, 2024-01-24:
Done

sanjay900, 2024-01-24:
did the slider start working in the tool?

Proofeek, 2024-01-24:
Nope

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14033400/firmware.zip)
this one should shove the slider values onto the whammy, does the whammy value in the tool update if you touch the slider?

Proofeek, 2024-01-24:
Yes it does

sanjay900, 2024-01-24:
can you get me the values from the `original value` box for pressing nothing, then G, R, Y, B, O and then GR, RY, YB, BO

Proofeek, 2024-01-24:
The values fluctuate within the range of a thousand
Nothing: 32000
G: 8400
R: 21000
Y: 38000
B: 49000
O: 65000
GR: 14500
RY: 27000
YB: 43000
BO: 54500

sanjay900, 2024-01-24:
pushing a 10.0.80 now, see if that works when it releases

sanjay900, 2024-01-24:
should be out now

Proofeek, 2024-01-24:
In version 10.0.80 when I save the settings, if there is a USB Host Inputs configuration, it crashes. If there is no USB Host Inputs, then it saves normally.

```
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/refs/heads/develop.zip; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip
Unpacking
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.10.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_sha.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\usbdsec.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\xsm3.c.o
Generating linkerscript C:\Users\kiill\AppData\Roaming\SantrollerConfigurator\Santroller\.pio\build\pico/memmap_default.ld
Compiling .pio\build\pico\libaa8\SPI\SPI.cpp.o
Compiling .pio\build\pico\libda2\USB-Host-Shield-20\AMBX.cpp.o
Compiling .pio\build\pico\libda2\USB-Host-Shield-20\BTD.cpp.o
Compiling .pio\build\pico\libda2\USB-Host-Shield-20\BTHID.cpp.o
In file included from src\shared\main\shared_main.cpp:1353:
src\shared\main\inputs/usb_host.h: In function 'uint8_t tick_inputs(void*, USB_LastReport_Data_t*, uint8_t)':
src\shared\main\inputs/usb_host.h:12:24: error: 'get_usb_host_device_tap_bar_type' was not declared in this scope; did you mean 'get_usb_host_device_type'?
   12 |     uint8_t tap_type = get_usb_host_device_tap_bar_type(i);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        get_usb_host_device_type
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1
========================= [FAILED] Took 12.83 seconds =========================

```

sanjay900, 2024-01-24:
oh shoot

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14033889/firmware.zip)

Proofeek, 2024-01-24:
This is the one that displays slider values as whammy. The problem remains the same.

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14033964/firmware.zip)

Proofeek, 2024-01-24:
It currently works like this (Xinput values that I see in the gamepad tester seem normal.):

https://github.com/Santroller/Santroller/assets/78957574/c3e4ea8f-9f70-42f3-8d59-c8bd8ec3a450

Proofeek, 2024-01-24:
Seems like everything is the same

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14034181/firmware.zip)

this one ive just disabled trying to work out if its a GH5 or a WT tap bar

Proofeek, 2024-01-24:
Now the icon only lights up when you touch the slider. The text "Green Red Blue Orange" appears when holding G or GR, in other cases "none"

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14034264/firmware.zip)

Proofeek, 2024-01-24:
This is in v10.0.82 with the latest firmware. Now the Xinput values fluctuate (along with the icon), whereas before everything was stable. The text "Green Red Blue Orange" appears when holding GR. 

https://github.com/Santroller/Santroller/assets/78957574/a37cb3fd-7f9d-4ed0-a053-6336c59befe7

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14034792/firmware.zip)

Proofeek, 2024-01-24:
Whammy is now whammy, everything else remains the same.

Proofeek, 2024-01-24:
If I create it myself, the frets are displayed correctly, but they still jitter (for example, none-blue if holding blue). This wasn't the case in this:

> [firmware.zip](https://github.com/Santroller/Santroller/files/14034181/firmware.zip)
> 
> this one ive just disabled trying to work out if its a GH5 or a WT tap bar

sanjay900, 2024-01-24:
oh, so it was working but it had jitter

Proofeek, 2024-01-24:
Each one

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14034931/firmware.zip)

and this one does the same?

Proofeek, 2024-01-24:
No, this one is ok

https://github.com/Santroller/Santroller/assets/78957574/699915ae-9fcd-4298-bd13-736467a67ac3

sanjay900, 2024-01-24:
in the tool as well?

Proofeek, 2024-01-24:
Like this

> It currently works like this (Xinput values that I see in the gamepad tester seem normal.):
> 
>  2024-01-24.10-31-05.online-video-cutter.com.mp4

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14034993/firmware.zip)

sanjay900, 2024-01-24:
its expected for the slider and whammy not to have the same values or even similar values, theres a good reason for that

Proofeek, 2024-01-24:
I mean that in this firmware, the axis 0 always shows the expected value when holding the position (except for G, which returns the value of GR). and it becomes zero only when I release the slider:
https://github.com/Santroller/Santroller/assets/78957574/949108a8-54a2-473f-b22e-6aa72db461b5

while here, it sometimes becomes zero when holding it (It's visible in the whammy values here that I am holding the slider.):
https://github.com/Santroller/Santroller/assets/78957574/705c2b1b-e623-48d2-a18b-ab49d368e862

sanjay900, 2024-01-24:
yeah right, did the new one make any difference at all

Proofeek, 2024-01-24:
The new one is the first video. G and GR return the same value.

sanjay900, 2024-01-24:
[firmware.zip](https://github.com/Santroller/Santroller/files/14035141/firmware.zip)
this one should make it so that G returns a different value

Proofeek, 2024-01-24:
No, it's the same

sanjay900, 2024-01-24:
I had a feeling it might be that

sanjay900, 2024-01-24:
guess the only important question left is does it work in game?

Proofeek, 2024-01-24:
Let me try

sanjay900, 2024-01-24:
And are you playing WT or GH5 or something else

Proofeek, 2024-01-24:
Everything is working flawlessly. WT

sanjay900, 2024-01-24:
perfect. want to check v10.0.83 just to make sure it saves everything correctly and then we can call this solved

Proofeek, 2024-01-24:
Whammy values are taken from the slider, everything else is working as intended!

sanjay900, 2024-01-24:
oh oops yeah ill fix that

Proofeek, 2024-01-24:
In v10.0.84, everything works as it should. Thank you for such a great tool!

closed 2024-01-24 by Proofeek