Santroller/Santroller #127: [Request]:

Proofeek, 2024-01-23:
### Please describe your request.

It would be great if Santroller could emulate Rock Band pro keys, allowing you to play with any USB MIDI keyboard by connecting it through USB host, like this:
[image]

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-01-23:
Cool idea but not sure when I'd get to it - midi input is a much more complicated thing than standard controllers

Moose1002, 2024-01-31:
Some sort of way to replicate a midi pro adapter using Santroller would be awesome! Hope one day you get around to exploring that avenue.

sanjay900, 2024-01-31:
It's one of those things on the list but I have no idea when it'll happen

dude98, 2024-02-10:
this could also allow people who have the xbox 360 version of RB3's pro keys to use it on ps3 without buying a wii or ps3 midi pro adapter.
you could probably use RPCS3's midi adapter emulation as a base for figuring out midi support.

sanjay900, 2024-02-10:
The main problem here is MIDI input and output, which RPCS3 doesn't actually really help with since thats on a USB layer and RPCS3 is much higher than that. I do want to implement it at some point, its just that there are more important things to get working first

sanjay900, 2024-05-23:
Implemented now

closed 2024-05-23 by sanjay900