Santroller/Santroller #128: [Issue]: Problem with drum multiplexer

goku1205, 2024-01-27:
### Please describe your issue.

I have a problem setting up a multiplexer for drums. The piezo sensors connected directly to the Raspberry Pico work correctly. However, when using the HP4067 multiplexer on separate channels, when tapping the drums, the buttons act as one. It seems like it's not separating the signal into channels.

### Version

v10.0.91

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-01-27:
Did you increase the minimum? This happens if you don't have the sensitivity calibrated correctly, as what your actually seeing is the pads picking up vibrations from the other pads being hit

goku1205, 2024-01-28:
I changed the minimum values, but the issue persists. It's possible that I made a mistake somewhere in the circuit during the creation. The Santroller project is really great because it breathes new life into my old non-functional drum set and guitar. Below, I'm providing the circuit diagram that I created.
[image]

sanjay900, 2024-01-28:
Looks right, just keep increasing the minimum even higher 
I presume the resistors are across red and black right?

goku1205, 2024-02-04:
There is a 1M ohm resistor connected with a red and black wire. I checked today, and when I hit the first piezo, it shows the same value on the second piezo.

goku1205, 2024-02-04:
[image]

sanjay900, 2024-02-04:
send a copy of `%apppdata%/SantrollerConfigurator/Santroller/include/config_data.h`

sanjay900, 2024-02-04:
alright nevermind i see the problem
i suspect it will work once you set everything up, the values in the tool are just wrong

sanjay900, 2024-02-05:
It is just a visual problem though - if you set a decent minimum itll actually work in game, its just not showing correctly in the tool.

sanjay900, 2024-02-05:
Fixed by v10.0.100

closed 2024-02-05 by sanjay900