Santroller/Santroller #129: [Issue]: pico guitar on ps2 not working

pillowyfish98, 2024-01-28:
### Please describe your issue.

im trying to get my guitar to work on my ps2 im trying ps2 controller inputs but it keeps asking about MOSI | MISO | SCK | Acknowledge | Attention pins i dont know how to find them out im just used to the auto pin mapping any help is appreciated 

### Version

10.0.91

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

PS2

pillowyfish98, 2024-01-29:
it was a kit for my wii wt guitar

sanjay900, 2024-01-29:
You'll need to contact whoever sold that to you. If they are doing things right they may even have a special build that preconfigures everything for you

sanjay900, 2024-01-29:
Wait

sanjay900, 2024-01-29:
Are you saying your trying to use this on a PS2?

sanjay900, 2024-01-29:
PS2 controller inputs isn't for that, PS2 controller inputs is for plugging a PS2 controller into the Pico.

What game?

sanjay900, 2024-02-01:
Did you ever get it working?

pillowyfish98, 2024-02-02:
no I didn't it wouldn't recognize my inputs

sanjay900, 2024-02-02:
And you have pademu?

sanjay900, 2024-06-09:
I went and purchased a PS2. Its working perfectly fine in GH2 after setting up PADEMU, ive updated the instructions to make things a bit more clear.
https://santroller.tangentmc.net/console_guides/ps2.html

closed 2024-06-09 by sanjay900