Santroller/Santroller #130: [Request]: PS3-style player LEDs

Proofeek, 2024-01-30:
### Please describe your request.

It would be great to have the option to set the Player LED setting not only to the player number but also to whether the player is number "" or higher. For example, to have the LED light up not only when the player is number 1 but also when are 2, 3, and 4, allowing the creation of a PS3-style player indicator. for example, for player 3, three LEDs light up, not just the third one.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-01-30:
Nothing is stopping you from just adding multiple leds and binding them all to the same set of leds

Proofeek, 2024-01-30:
Oops, sorry, I never had a ps3, I thought they worked that way, but anyway, when I bind multiple player numbers to one LED, it doesn't light up at all.

sanjay900, 2024-01-30:
Damn, that's unfortunate, I'll have a look later

sanjay900, 2024-01-30:
Ah yeah i can see why that wouldn't work, ill have to do some better logic for that to make this sort of thing work

dude98, 2024-02-14:
I'm gonna add a small thing here to say that a Multi Light Player Indicator is something the Nintendo Switch does.
Here's the player light list taken straight from nintendo's website: (1 is On, 0 is Off, viewed horizontally)
P1 - 1 0 0 0
P2 - 1 1 0 0
P3 - 1 1 1 0
P4 - 1 1 1 1
P5 - 1 0 0 1
P6 - 1 0 1 0
P7 - 1 0 1 1
P8 - 0 1 1 0

sanjay900, 2026-11-08:
this is in now

closed 2026-11-08 by sanjay900