Santroller/Santroller #131: [Request]: idle shutdown on bluetooth

mandryd, 2024-01-31:
### Please describe your request.

First off, thanks for your amazing work on this tool.
  
I currently am using a Bluetooth guitar powered with a tp4056 module and an 18650 battery.  I have a power switch to turn the whole device off but was wondering if there could be an idle shutdown if no keypress in a set time to save power if I forget to turn it off?  Thanks!

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-01-31:
Its on the list but im not sure when ill get to it

mandryd, 2024-02-02:
Awesome, no rush :)

repvik, 2024-10-21:
How about a deep sleep for the entire Pico after X minutes since last activity? I believe the Pico can wake up from deep sleep when a GPIO state change is detected. 

Also, an alternative for controllers like this: https://www.printables.com/model/779720-les-paul-controller-for-guitar-heroclone-heroyarg is pulling a GPIO high for a bit then low. That can turn off the LiPo shim cutting power to the Pico completely.

mandryd, 2024-10-21:
That's my guitar design :).  I added the lipo shim support after my initial request here.
A sleep is an interesting idea also, especially combined with a later shutdown.  My main concern is just forgetting to turn it off and draining all the battery because my kids can be absent minded about it.

repvik, 2024-10-21:
Fantastic controller!
I've printed two and I'm waiting for more hardware to wire them up w/shim. I suspect I will almost always forget to turn them off. 
Powersaving is the one thing missing.

sanjay900, 2025-01-25:
Santroller v10.1.345 has a sleep mode timer now, that puts the pico into a deep sleep, and then you can configure a pin to wake it back up.

closed 2025-01-25 by sanjay900

mandryd, 2025-01-26:
Awesome!! Thanks!