Santroller/Santroller #132: [Request]: Wii Compatibility without using Fakemote

SirRob24, 2024-02-03:
### Please describe your request.

Target User: This would be for people who don't have a modified Wii but wish to be able to play GH and RB with their Santroller.

Goal: Playing GH on the Wii with a non-Wii guitar without the use of Fakemote via an additional output mode alongside a Wiimote extension cable/Wiimote socket that's connected to your microcontroller. With this mode you'd be able to manually set which pins you'd put the clock and data on. Be able to use the USB or Wii output without reprogramming.

The General Idea: So say I have my Santroller setup. Lets say it's an original GH SG. I can plug it in via USB and play Clonehero on my PC. I can then plug it in USB to my Wii to play Rock Band because I set it up as RB guitar. "I want to play Guitar Hero on my Wii instead now." No problem, I now plug into the Wiimote using the additional port I've installed. Now the Wii sees it as a GH guitar and I can bootup GH Aerosmith and play until my fingers fall off.

Just an idea that's been floating around my head so I don't know how useful or necessary something like this would be.

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-02-03:
RB works over usb, I don't think the Wii is worth adding support for

sanjay900, 2024-06-01:
Should be a thing in the next release.

sanjay900, 2024-06-01:
[image]

only thing to be mindful of is when this is released you can never plug in usb and the extension at the same time

sanjay900, 2024-06-01:
It's beta at the moment as I haven't been able to test every extension yet but guitars seem to work. Have just started the release process.