Santroller/Santroller #133: [Issue]: Clarify power details for Pico W in documentation

Soundtoxin, 2024-02-03:
### Please describe your issue.

I'm working on direct-wiring some Rock Band 2 Drums to a Pico W with the hopes of having both wired and wireless functionality and easily swapping between them.
If I have the batteries (3 AAs) in the back cover slots wired to the Pico W and then I decide I want to play wired, is it safe to plug in to USB with the batteries still in/powered on, or do I need to remove the power from the batteries first? (similarly would it be safe to hook up to a computer for configuring while the batteries are in there?) I haven't changed any actual wiring yet as I'm still in the investigation stage and trying to figure out what connects where, but I was going to use VSYS. There doesn't seem to be any power switch on the controller by default.

### Version

10.0.96

### What operating system are you using?

Other

### os-other

N/A

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

Soundtoxin, 2024-02-03:
I found this in the Pico docs:

>The simplest way to safely add a second power source to Pico is to feed it into VSYS via another Schottky diode (see
Figure 16). This will 'OR' the two voltages, allowing the higher of either the external voltage or VBUS to power VSYS, with
the diodes preventing either supply from back-powering the other. For example a single Lithium-Ion cell* (cell voltage
~3.0V to 4.2V) will work well, as will 3×AA series cells (~3.0V to ~4.8V) and any other fixed supply in the range ~2.3V to
5.5V.

Does this mean I'll need to add a schottky diode between the batteries and the Pico W to make switching between power sources on the fly safe? If so, that should probably be in the documentation either on the [bluetooth page](https://santroller.tangentmc.net/tool/bluetooth.html) or one of the other wiring guide pages.

sanjay900, 2024-02-03:
Yeah I probably should add that, I don't really have much on Bluetooth besides programming, never really got around to battery power docs

sanjay900, 2024-02-04:
I've added that now

closed 2024-02-04 by sanjay900

Soundtoxin, 2024-02-04:
Awesome! Thanks.