Santroller/Santroller #134: [Issue]: Analog Multiplexer inputs on Rock Band 2 drum controller not working

Soundtoxin, 2024-02-07:
### Please describe your issue.

I've direct-wired a Rock Band 2 drum with a [16 channel multiplexer](https://www.sparkfun.com/products/9056) and Pico W. I think I've done something wrong and maybe the documentation could be improved if so.

First, the problematic behavior I'm seeing is that pressing the drum pads does nothing, but also the pads and cymbals as I've programmed them in Santroller all show constant activity in the 5k range with some jittering up and down, their icons show constantly lit up. So it might think they're all being pressed for some reason. I have tried adjusting min/deadzone but since firmer hits don't register, I think I have bigger problems.

As for the wiring, I've got the drum pads (via test pads near their solder points on the PCB) on C3, C2, C1, C0 of the multiplexer. S3, S2, S1, S0 are wired to GP21, GP20, GP19, GP18, respectively.

There's a line in the docs that is a bit ambiguous:

>Wire COM on the multiplexer to an analog pin on your Pi Pico

I initially interpreted COM to mean common ground, so I used the GND pin on the multiplexer (nothing says COM on the actual breakout board I have, store page linked above with pics). I later noticed on the store page the following:

>This chip is like a rotary switch - it internally routes the common pin (COM in the schematic, SIG on the board)

This makes it sound like COM means SIG, not GND. So I moved my wire from GND to SIG, still going to the same spot on the Pico W, GP26_A0, the first analog pin. I did not notice any change in behavior at all from moving that wire, oddly... So whatever the issue is, that didn't seem to do anything. Would still like clarification on whether I should be using SIG or GND, and maybe a note added to the drum wiring page. Hopefully I didn't fry something using the wrong one. My problem may be elsewhere anyway.

I've got a GND test pad on the drum PCB (T28) wired to AGND on the Pico W for my ground connection between those boards.

My kick pedal works, but that's a digital input going right from the drum PCB (T14) to the Pico W (GP17), so if the problems are with the analog multiplexer and how it's connected, that doesn't tell me much other than that the Pico W works. So far I haven't wired up the ports for the cymbals (T3, T4, T5) since I don't have the actual cymbals to test with anyway, but I did configure them in the software since they'll just be more analog multiplexer inputs. I've got both kick pedals configured but only the main port is wired up since I won't be putting the back cover back on for a while and the other port is on there. I haven't wired up the dpad or face buttons because their test points seem to be on the other side of the board and I'm not sure if it'll fit together again if I have wires going under there. I'm guessing they'd work okay since they're just digital inputs.

I also have some questions/concerns about the software settings for analog multiplexer inputs. I seem to have to specify the analog pin and the S0/S1/S2/S3 input pins for every drum pad and cymbal even though these are the exact same for every drum pad (just the channel on the multiplexer changes between them). This made me worry I was doing something wrong, because it seemed silly to have to keep selecting the same things repeatedly. Should these not all be the same after all?

So far I have just configured this as a wired Rock Band drum set, I have not hooked up battery power or connected to a receiver or anything. I wanted to get it working in its most basic form first.

I have double-checked my wiring to make sure the signal is getting across the wires and that I don't have continuity with adjacent pins (accidental shorts). Everything appears to be fine so I'm at a bit of a loss. I could've missed something of course. I'm hoping you have some idea of what may be wrong here. Also, sorry to say I do not use Discord (I see you ask that in a lot of other issues), but if you could help me out at all here in this issue it would be much appreciated, and maybe someone else will benefit from what we figure out in the future as well.

### Version

10.0.102

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-02-07:
They do have to be specified every time - someone might want to use multiple multiplexers or something.

I'll update the line in the docs about COM - it seems different multiplexers use different naming standards for the pins

Do the numbers change at all when you hit the pads?

Soundtoxin, 2024-02-08:
No change at all from hitting the pads as far as I can tell.

sanjay900, 2024-02-08:
Then i gotta assume you have something wired incorrectly, Only other thing you can maybe try is cutting all the traces going to the original chip on the old drum brain.

Soundtoxin, 2024-02-08:
I'd like to avoiding cutting traces if possible. Everything else up to this point has been reversible, which I kinda liked. Though I don't have the dongle to use this thing stock anyway. Just feels less like I'm ruining this old hardware in limited supply that way.

Since all drum pads exhibit the same behavior I figured it was some problem with how the multiplexer is hooked up.

Is there any reason the board wouldn't work without battery power (but while plugged in)? 

Could I determine if the chip in the middle is causing an issue before I cut it? Have you actually had to cut traces on a drum brain before?

I've noticed if I short a drum pad signal pin to GND on the drum PCB that the value in the configurator drops. The constant value is 5400 or so and when shorting those pins it goes down to 3900. This change appears to affect multiple pads, probably all of them at once. For sure both red and green when I was shorting pins for red. Does this hint at where the wiring is potentially wrong?

I also notice if I short the green pad wire on the multiplexer to SIG in the multiplexer with a small wire that values drop to below 1000, around 450.

I don't notice a change trying to short the other colors, so I may touch up the soldering on those points.

sanjay900, 2024-02-08:
One thing you could try if you can easily get your hands on resistors, is to just totally desolder the wires going to the drum pads away from the board and just follow the instructions for soldering a pad from scratch instead of the modding instructions. I cut the traces on my board myself iirc

Soundtoxin, 2024-02-08:
Yes, latest version of the tool as of my original issue posting. I'm probably one or two behind now. Here are wiring pictures.

[image]

[image]

Looks a bit messy as I didn't remove all the stock cables yet.

I will consider the resistor route or cutting the traces I guess. It makes sense that the original chip could be causing issues.

sanjay900, 2024-02-08:
honestly if its like that you might just find it easier to desolder that little board with the blob on it

sanjay900, 2024-02-10:
FYI there are some multiplexer improvements in v10.0.110

Soundtoxin, 2024-02-10:
Thanks for the heads up. I haven't gotten around to desoldering that chip yet, but hoping to give it a go in the next few days, or if it's troublesome I'll probably just try to cut the traces after all. Is there any point trying the new release before the old microcontroller is out of the picture?

sanjay900, 2024-02-10:
Worth a shot maybe tbh, though wait for v10.0.111

Soundtoxin, 2024-02-10:
On v10.0.111 I am consistently getting hanging/Not Responding right after clicking Configure with my device plugged in.

sanjay900, 2024-02-11:
Can you send a copy of `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`
it will only show up after hitting configure

sanjay900, 2024-02-11:
> [image]

I just looked at this image again, it looks like nothing is connected to VCC, that should be going to 3v3 on the pico

Soundtoxin, 2024-02-11:
I didn't see a config_data.h, just a config.h in there. Thanks for noticing the wiring. I don't know if the guide was missing the info about wiring VCC before or if I just missed it. I'll get that wired up.

sanjay900, 2024-02-11:
Means you havent hit the save button yet, that file is generated when you save the config to the device

Soundtoxin, 2024-02-11:
Yeah, I have the config from a previous version, it hangs as soon as I hit Configure on the main screen, I see the controls but can't even scroll, so I don't think I'd be able to hit save either. I've got VCC wired up now, so I'll see if an older version of the software works now or not. I could probably also go into bootsel and totally wipe my config, but then I gotta enter a bunch of stuff again.

Soundtoxin, 2024-02-11:
Drum pads still don't work on v10.0.102 with VCC wired, so back to the plan of trying to desolder the old chip, I suppose.

sanjay900, 2024-02-11:
What happens if you wire it straight to an analog pin

Soundtoxin, 2024-02-11:
I tried holding a dupont cable to the test pad for the red drum pad and sticking the other end in the ADC2 hole, then hitting the drum pad, and I didn't notice a difference, but it's still also hooked up to the multiplexer and everything, if that matters.

sanjay900, 2024-02-11:
I also just wonder if the PCB for this kit is just incompatible with PCB wiring like this honestly

Soundtoxin, 2024-02-11:
That might indeed be the case. I got the old microcontroller desoldered and still the drum pads don't wanna work.

[image]

sanjay900, 2024-02-11:
You may just have to grab some resistors and follow the alternate instructions that dont use the original pcb

sanjay900, 2024-02-11:
when it comes out release 112 will probably stop the crashes

Soundtoxin, 2024-02-11:
Nice, will try the new version in a bit then. After some sleep I realized when I tried the red drum pad straight to an analog pin earlier, I forgot to change the setting in the software from a multiplexer input to a regular analog one, so of course it wouldn't work. I will do some more experimenting later. The fact that the digital kick pedal can work through the old PCB makes me think the drum pads should be able to also, but maybe it's not that simple.

Soundtoxin, 2024-02-11:
I can confirm no more hanging/crashing on v10.0.112.
The red drum pad going straight to ADC2 might be working, it's a little hard to tell. "Find Pin" selected the right one for me, there was over 60k for the value without pressing anything. I cranked the min way up and then tried to do hard flicks right at the center of the drum pad while holding the wire in place. I was either getting small changes in value occasionally or my wire moving around or the normal amount of value jittering tricked my eyes. Hard to say. It's tricky to balance with the PCB that wants to fall out and so on. Later I may just properly solder the red pad wire to an analog pin and tape the PCB down so I don't have go set it pad-down on the table. I haven't been able to get the silicone membranes for the dpad and buttons back into place, so I haven't re-screwed down the PCB since initial disassembly and cleaning.

Soundtoxin, 2024-02-11:
Took the membranes out, screwed PCB in somewhat, soldered red drum pad wire to ADC2 instead of the multiplexer.

I'm getting really weird behavior now. gamepad-tester shows the right face button and the left analog stick always dark, B1 and B11 lit up. Without touching it, occasionally there's a flash and some other things are "pressed". Including right sticv and dpad down. B2, B5, B13.

Something is working somewhat but there's clearly a problem somewhere. Lifting the drum off the table doesn't seem to stop it thinking stuff is pressed.

Soundtoxin, 2024-02-15:
Thanks, I will look it over. So do you recommend just totally removing the original PCB then? My main concern with that route is that it becomes a lot harder to get a working dpad, face buttons, start/select/guide, etc. if I ditch the original PCB. I did order some resistors from Amazon now so I should be able to attempt the other method soon.
I was thinking if necessary I could wire up some keyboard switches or tactile switches to digital pins on the Pico to replace the controller buttons, but then I'd need some way to keep them in place somewhere. I don't think that'd all fit in the original case or work with the old buttons. I'd probably need a 3D-printed holder or something that'd hang from the front of the drum.

sanjay900, 2024-02-15:
Use the original PCB for buttons, but then bypass it for the pads

Soundtoxin, 2024-02-18:
>Solder the black wire from the piezo to ground.

>Solder a 1Mohm resistor between the red and black wires on the piezo.

If I hook to the GND I've been using so far, isn't that still going through the Drum PCB then? Is it more likely to be fine if just the GND goes through that and not the signal? With the digital stuff still going through the board (including GND) and all the GNDs being connected it sounds like GND-wise it'll be the same as by default (except that maybe it sounds like I need to put a resistor in-line with it).

I'm having trouble picturing where the resistor goes and how many are needed. I was imagining it'd go between the signal wire and the multiplexer. Does each piezo need two resistors then? One per wire (GND getting its own)? I'm assuming you don't connect both to the same resistor or it'd be shorted on.

sanjay900, 2024-02-18:
One resistor, between ground and signal wire going from the piezo to the piezo. You should be fine to reuse ground from the main pcb

sanjay900, 2024-02-26:
If you can't get the multiplexer working, the docs have been updated with an alternative where you just wire drums up to digital pads, which will work for the games.

Soundtoxin, 2024-02-26:
Ah, cool to have that option. Thanks. Haven't had the chance to rewire the drum pads yet, been a bit busy. Will update you when I next work on it.

sanjay900, 2024-03-10:
I finally put together a board with a multiplexer (not a drum mind you but something else) and I was able to confirm the multiplexer code does work

sanjay900, 2024-05-04:
If you want to get the multiplexer working, check the updated drum guide

Soundtoxin, 2024-05-04:
Thanks, I'll pick some up!

sanjay900, 2024-05-04:
Even got the slower multiplexer working too, i think the high voltage from the piezo was just really confusing the multiplexer, but a zener diode is enough to clamp that

closed 2024-05-24 by sanjay900