Santroller/Santroller #137: [Issue]: linux

rebornshadow134, 2024-02-17:
### Please describe your issue.

installation on Ubuntu 22.04.4 LTS fails every time

### Version

v10.0.126

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-02-17:
fails how?

rebornshadow134, 2024-02-17:
fails to install and says not not supported, using Ubuntu software installer and tried gnome installer same error.

sanjay900, 2024-02-17:
It is just an executable, set it as executable and run it, there is nothing to install

rebornshadow134, 2024-02-17:
I've tried just running it as an executable nothing ever loads up

sanjay900, 2024-02-17:
do you see anything in the console?

rebornshadow134, 2024-02-17:
no

sanjay900, 2024-02-17:
unplug your guitar and see if that makes a difference

rebornshadow134, 2024-02-17:
still nothing, i get spinning mouse but nothing more.

sanjay900, 2024-02-17:
Im gonna go download a copy of 22.04.4 and see if somethings just broken in that build being that its a daily build and not stable

rebornshadow134, 2024-02-17:
okay thanks:)

sanjay900, 2024-02-17:
why are you using 22.04.4 anyways?

rebornshadow134, 2024-02-17:
for some reason my PC updated earlier when i switched to Ubuntu Pro

rebornshadow134, 2024-02-17:
i didn't even know that it wasn't stable, its been a hot minute since i used Ubuntu.

sanjay900, 2024-02-17:
Somehow your on https://cdimage.ubuntu.com/jammy/daily-live/current/ which is 22.04.4
Since https://releases.ubuntu.com/jammy/ is still 22.04.3

rebornshadow134, 2024-02-17:
reverting back to jammy 22.04.3 ill keep you posted

sanjay900, 2024-02-17:
sounds good

rebornshadow134, 2024-04-30:
> Did you ever get this going? Recently things were swapped to appimages, so
> that might work better?

I did finally get it to work, had to change some settings in console.

closed 2024-04-30 by sanjay900