Santroller/Santroller #138: [Issue]: RB kick pedal doesn't work on xb1

jrzeszutek, 2024-02-21:
### Please describe your issue.

Hi @sanjay900, first of all thanks for an amazing job you're doing here!
I have a Rock Band 1 drum set, which I wired with Pico. All works perfectly fine on PC, e.g. in Clone Hero, but when I'm trying to use it in Rock Band 4 on my XB1, for some reason kick pedal doesn't work. Other drum pads are recognized correctly. I suspect there is some issue in buttons mapping. 

I figured out that the game maps the kick pedal to RB button on Xbox Controller, but it doesn't work via USB Inputs:
[image]
On the other hand, other buttons (red, green, blue, yellow) are mapped correctly:
[image]

### Version

v10.0.133

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-02-21:
Question, which multiplexer did you use?

jrzeszutek, 2024-02-21:
TBH - none :) I followed wiring steps and connected RB kick pedal to digital pin: https://santroller.tangentmc.net/wiring_guides/drum.html#wiring-steps

When it comes to drum pads, since it's RB1 kit with no velocity support, I assumed it should work on digital pins and it does. The only problem I'm facing is kick pedal. Like I mentioned before - only on XB1. In Clone Hero PC works fine.

sanjay900, 2024-02-21:
The RB1 kits are analog though when it comes to direct wiring, so if you do want analog data you can wire it with a multiplexer to get that functionality. 
Could you send a copy of the file `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`

sanjay900, 2024-02-21:
Hopefully, [v10.0.135](https://github.com/Santroller/Santroller/releases/tag/v10.0.135) will fix the kick for your direct wired kit.

Due to how the usb host stuff works, the gamepad wont pass through kick by default

jrzeszutek, 2024-02-21:
> The RB1 kits are analog though when it comes to direct wiring, so if you do want analog data you can wire it with a multiplexer to get that functionality. Could you send a copy of the file `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h` I am going to add wiring to digital pins to the guide, thanks for that

I didn't knew about that. I already ordered 4051 multiplexer, so I'll give it a try. Thanks!
Below the file you have asked for:
```
#define CONFIGURATION {0x1B,0x19,0x4,0x8,0x8E,0x94,0x2E,0xE2,0xDE,0xFF,0x7F,0xEF,0xEE,0xF8,0x63,0xB7,0xD6,0xC0,0x28,0xC,0xC3,0x4E,0x44,0x8F,0xA3,0x80,0x3,0xC,0xA3,0x4E,0x46,0x71,0x9B,0x7A,0x25,0x9,0x7B,0x71,0x53,0x6B,0x83,0x4E,0x83,0x7E,0x28,0x34,0x26,0x26,0x3C,0x96,0x6F,0x1C,0x40,0x1,0x75,0xA9,0x66,0x92,0xE6,0x36,0x86,0x1D,0xAF,0x37,0xC1,0x4E,0xD0,0xE8,0x15,0x3F,0xE6,0xB5,0x65,0xE7,0xA1,0x35,0xAD,0xE0,0x59,0xF3,0x59,0xEA,0x45,0x23,0x23,0xE2,0x97,0x92,0xE3,0x98,0x57,0xFC,0x42,0x88,0x9F,0x2F,0xCE,0xAF,0x45,0xE8,0x4B,0xC9,0x6B,0x6B,0xF,0xCB,0xE5,0x43,0x9B,0x79,0x8C,0x60,0x59,0x17,0xE2,0x2F,0x25,0xC7,0x80,0x80,0xC5,0x6,0x29,0xF4,0xDC,0xED,0xB3,0xA1,0x49,0xCF,0x24,0x88,0x1F,0x16,0xF9,0x87,0xAF,0x7A,0xD3,0x9F,0xAA,0x95,0xBF,0x69,0x19,0x4B,0x3D,0xE9,0x94,0x47,0x67,0x5C,0xA7,0x21,0xCC,0x71,0x5,0xEB,0xA6,0xA0,0x5F,0x28,0x71,0x64,0xA,0xA1,0xA8,0xE4,0x36,0x63,0x47,0x58,0x18,0x53,0x8C,0x99,0x94,0x85,0x19,0x94,0x8D,0x39,0x94,0x83,0x65,0x29,0xD,0xCB,0x51,0x3A,0x96,0xA7,0xC,0xAC,0xA0,0x9B,0x4E,0xF5,0xA1,0x95,0x90,0x79,0x7A,0xDE,0x5D,0xDB,0xE2,0x1E,0xFA,0x9C,0x19,0xA6,0x73,0x16,0xB2,0x2A,0x94,0x95,0x2,0x63,0xB2,0x15,0xE5,0xC,0xF7,0x10,0x9C,0x11,0x46,0x9C,0x31,0xC6,0x1C,0xD,0x33,0x38,0xBA,0x64,0x3E,0x6,0x59,0x4A,0xDF,0x67,0xF8,0x51,0xC1,0xC,0x36,0x76,0x76,0xEA,0x15,0xAD,0xBB,0x1A,0x7,0x2,0x46,0xC2,0xDF,0x40,0x70,0xEE,0xC8,0x8}
#define CONFIGURATION_LEN 266
#define SWAP_SWITCH_FACE_BUTTONS true
#define WINDOWS_USES_XINPUT true
#define RPCS3_COMPAT true
#define XINPUT_AUTH true
#define INPUT_QUEUE false
#define POLL_RATE 0
#define WT_SENSITIVITY 5
#define INPUT_DJ_TURNTABLE_POLL_RATE 10000
#define INPUT_DJ_TURNTABLE_SMOOTHING false
#define LED_BRIGHTNESS 31
#define ABSOLUTE_MOUSE_COORDS false
#define ARDWIINO_BOARD "pico"
#define EMULATION_TYPE 0
#define DEVICE_TYPE 6
#define USB_HOST_STACK true
#define USB_HOST_DP_PIN 0
#define DIGITAL_COUNT 25
#define LED_COUNT 0
#define LED_COUNT_PERIPHERAL 0
#define LED_COUNT_STP 0
#define LED_COUNT_PERIPHERAL_STP 0
#define ADC_PINS {}
#define ADC_COUNT 0
#define TICK_SHARED \
    if (((sio_hw->gpio_in & (1 << 22)) == 0 )) {\
        debounce[4]=2; \
    }if ((usb_host_data.x )) {\
        debounce[5]=2; \
    }\
    if ((usb_host_data.a )) {\
        debounce[6]=2; \
    }\
    if ((usb_host_data.b )) {\
        debounce[7]=2; \
    }\
    if ((usb_host_data.y )) {\
        debounce[8]=2; \
    }\
    if ((usb_host_data.start )) {\
        debounce[9]=2; \
    }\
    if ((usb_host_data.back )) {\
        debounce[10]=2; \
    }\
    if ((usb_host_data.guide )) {\
        debounce[11]=2; \
    }\
    if ((usb_host_data.dpadUp )) {\
        debounce[12]=2; \
    }\
    if ((usb_host_data.dpadDown )) {\
        debounce[13]=2; \
    }\
    if ((usb_host_data.dpadLeft )) {\
        debounce[14]=2; \
    }\
    if ((usb_host_data.dpadRight )) {\
        debounce[15]=2; \
    }\
    if ((usb_host_data.kick1 )) {\
        debounce[23]=2; \
    }\
    if ((usb_host_data.kick2 )) {\
[... 804 lines cut ...]
    gpio_set_dir(21,false);\
    gpio_set_pulls(21,true,false);\
    gpio_init(18);\
    gpio_set_dir(18,false);\
    gpio_set_pulls(18,true,false);\
    gpio_init(19);\
    gpio_set_dir(19,false);\
    gpio_set_pulls(19,true,false);\
    gpio_init(20);\
    gpio_set_dir(20,false);\
    gpio_set_pulls(20,true,false);\
    gpio_init(22);\
    gpio_set_dir(22,false);\
    gpio_set_pulls(22,true,false);\
 \
 

#define INPUT_DIRECT
#define INPUT_USB_HOST
```

> Hopefully, [v10.0.135](https://github.com/Santroller/Santroller/releases/tag/v10.0.135) will fix the kick for your direct wired kit.
> 
> Due to how the usb host stuff works, the gamepad wont pass through kick by default
I will give it a try and let you know about the results. Thanks again!

sanjay900, 2024-02-22:
was that config_data.h from 10.0.135?

sanjay900, 2024-02-22:
Try `[v10.0.140](https://github.com/Santroller/Santroller/releases/tag/v10.0.140)`

closed 2024-02-22 by jrzeszutek

sanjay900, 2024-02-22:
You wouldn't have been able to find the issue anyways, it was in the GUI side of the codebase which isn't open source

jrzeszutek, 2024-02-22:
> You wouldn't have been able to find the issue anyways, it was in the GUI side of the codebase which isn't open source I had just forgotten to define mappings for the pedal on XB1, because it uses a totally different set of code since its mappings are totally unique in comparison to the other consoles lol

Well... Just regular Microsoft stuff, it doesn't surprise me at all lol

sanjay900, 2024-02-22:
Actually its for a good reason, the other consoles force every controller to use a controller-like layout but the xb1 lets controllers just define totally custom layouts
so where on other consoles they had to do dumb things like combine some velocities and put them on the sticks, the xb1 just has a value for each drum pad lol

jrzeszutek, 2024-02-22:
> Actually its for a good reason, the other consoles force every controller to use a controller-like layout but the xb1 lets controllers just define totally custom layouts so where on other consoles [...]

Now THAT surpises me actually. Thanks for explanation!