Santroller/Santroller #139: [Request]:  LED disable on idle.

al3ph, 2024-02-21:
### Please describe your request.

I've set my frets to have unpressed colours, but this means I have have to disconnect the guitar when not in use (to stop the frets glowing all the time).
It's be handy for the fret LED's (or in fact any LED's) to go to sleep if the guitar has detected no activity for N seconds.

Cheer!

### What operating system are you using?

Linux

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2025-01-25:
Santroller v10.1.345 has a configurable timer to turn off LEDs after a given idle period

closed 2025-01-25 by sanjay900

al3ph, 2025-01-25:
Nice one !
Cheers!