Santroller/Santroller #142: [Issue]: Controller does not show as guitar hero controller under linux

NikonUK, 2024-02-26:
### Please describe your issue.

so im trying to run guitar hero 3 under arch. the game runs absolutely fine but the controller acts as if its a standard xbox 360 controller instead of a guitar hero controller. im unsure if its a driver issue, a wine issue, your software issue. but either way i just thought i should bring this up and hope that you can help fix it?

### Version

latest (cant find the version number but i installed it not even an hour ago)

### What operating system are you using?

Linux

### os-other

Arch

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

NikonUK, 2024-02-26:
this makes the game unplayable as the game treats the controller with different mappings that i havnt got a clue are

NikonUK, 2024-02-26:
and just to add, it is set as a guitar hero  xinput controller in your software, not a gamepad

sanjay900, 2024-02-26:
Yeah, thats just how linux works, because wine doesn't know how to emulate xinput controller types.
Heres the fix:
```
Wine doesn't like controllers  that don't fit the standard XInput layout, which none of ours do as we expose controllers with all buttons. 
Perform the following actions to disable wines automatic mapping, which will make the controllers work properly

Open regedit in your WINEPREFIX
Navigate to HKEY_LOCAL_MACHINE\System\CurrentControlSet\Services\winebus
Create a new DWORD named "Enable SDL" with value 0
Create a new DWORD named "Map Controllers" with value 0
```
Then download a copy of x360ce *v3* and place it next  to the game, along with the following ini file
[x360ce.ini.zip](https://github.com/Santroller/Santroller/files/14410881/x360ce.ini.zip)

closed 2024-02-26 by sanjay900