Santroller/Santroller #144: [Issue]: whammy calibration mapping doesn't work, raspberry pi pico the correct values or dont read at all

rhopecity, 2024-02-27:
### Please describe your issue.

I recently took an old ps2 kramer that didnt have a dongle and wired it to run through a raspberry pi pico. I was able to map all the buttons no problem and everything works perfectly in game on pc gh games and even with my ps5 on rock band 4, everything except for whammy. Whammy is recognized perfectly by santroller however is being miss read in control panel and in game. It seems like the calibrated value is not being read by windows and instead its reading the original value of the potentiometer. That or it the calibrated signal is just not being mapped onto the controller the same way it reads in santroller. Im not sure thats why I'm asking you LMAO. This isn't a problem on clone hero since the whammy calibration is a bit more advanced and the whammy can be mapped properly, but on ghwt:de and on console the whammy bar does not function because of this.

and yes i have triple checked and everything is soldered correctly and strongly, the potentiometer gets read by the computer, its just not getting mapped correctly by santroller to be able to used in the guitar hero and rockband games.

https://www.youtube.com/watch?v=CjcWwBgLSko
Above is a video showing the calibrated value being written to the pi pico (the new brains of the kramer i modded) and showing the difference in readings between santroller and windows/in game. Reads perfect in santroller, reads incorrectly in windows and in game. And yes i have already tried guitar configurator, when i used it the whammy wasnt read at all by windows.

issue occurs with both Xinput and HID Game controller modes.

### Version

v10.0.159, but prior versions also have the same bug

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-02-27:
Are you sure you wired that correctly? Those raw values seem very very odd.

rhopecity, 2024-03-12:
Sorry for taking two weeks to reply, work had me busy and I was unable replace the pot for a little while, but, you were correct in thinking the pot was broken, it was LOL. Working great with a new pot thank you for your software!

closed 2024-03-12 by sanjay900