Santroller/Santroller #145: [Request]: Dancepad Compatibility

dude98, 2024-03-02:
### Please describe your request.

with how easy it is to wire and configure this, would it be possible to make dance pads an option if they're not already? i can't find anything on how to do it, but i don't see why, since being able to convert a ps2 pad to usb or any DIY/USB pad to work on both console and pc with one adapter would be awesome.

microcontroller request and how it's programmed is N/A because it'd be a new configuration for all of them, as well as support PS2 Adapter Wiring and Direct Wiring.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-03-02:
what console do you want this to work on

dude98, 2024-03-02:
the consoles that have DDR games you would use this with are PS2 using OPL's PADEMU, PS3, X360, and if possible Original Xbox (not a requirement, just telling you since the original xbox controller ports are just USB ports internally)
The Buttons that my X360 Dance Pad Uses are DPad for the Arrows (Left and Right or Up and Down work, but i'm not fully sure how) the face buttons are on the corners of the pad (A is the Top Right Button, B is Top Left, X is Bottom Right and Y is Bottom Left, However for PS2 mats A and B are flipped so Circle is Top Right and Cross is Top Left Because in Japan the Confirm Button is Circle) and Start and Select/Back are above A and B, with Start usually on the Right and Select on the Left.

Again, the ability for this to work on Original Xbox is not entirely necessary since it'd just be for dance pads and regular controllers (No Guitar Hero or Rock Band games were on Original Xbox) but it'd be nice to have the ability to play DDR Ultramix 3's Doubles since i only have 1 Original Xbox Dancemat.

sanjay900, 2024-03-02:
if the games just want a standard gamepad, then it should just work

sanjay900, 2024-04-08:
Original xbox should work now as well

sanjay900, 2024-05-24:
I don't see any reason why you couldnt do dance pads with the firmware as it is now, you would just set it to gamepad and everything should just work.

closed 2024-05-24 by sanjay900