Santroller/Santroller #146: [Issue]: Using Wii USB Rockband Drums on PS4 with Bluetooth

RE-Gaming, 2024-03-02:
### Please describe your issue.

I have setup the Wii Rockband Drums to work via USB (Drums --> PICO USB Input on GP2 and GP3 --> PICO USB Port --> PS4), this works perfectly fine. 

My next step was to reprogram this as a Bluetooth setup so that I no longer need a wire across the living room floor. I got this setup and working (Drums --> Bluetooth transmitter PICO USB Input GP2 and GP3 powered by USB power bank) connected to (Bluetooth receiver PICO --> PS4)

This mostly works, every pad and button works except the kick pedal. I've verified with a multimeter that the kick pedal is working.

### Version

10.0.161

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2024-03-02:
I've likley just missed the bt mappings I'll take a look in a bit

sanjay900, 2024-03-13:
Does the kick pedal light up in the tool at all?

sanjay900, 2024-03-13:
the kick not working should be fixed by v10.0.173
no idea why that disconnection error happens, but its unrelated

closed 2024-03-13 by sanjay900

sanjay900, 2024-03-14:
Which wii rb drum kit have you got?

RE-Gaming, 2024-03-14:
I believe it is rockband 1

sanjay900, 2024-03-14:
How bizarre - I was under the impression that Wii and PS3 kits are almost identical but PS3 rb1 kits don't work properly at all

sanjay900, 2024-03-14:
Could you run this and get me a descriptor dump from it?
https://www.thesycon.de/eng/usb_descriptordumper.shtml

RE-Gaming, 2024-03-15:
[DescriptorDump_Harmonix_Drum_Controller_for_Nintendo_Wii.txt](https://github.com/Santroller/Santroller/files/14609869/DescriptorDump_Harmonix_Drum_Controller_for_Nintendo_Wii.txt)

RE-Gaming, 2024-03-15:
I also have a wii keytar is that is of interest for a descriptor dump

sanjay900, 2024-03-15:
Yeah that could be useful

RE-Gaming, 2024-03-15:
[DescriptorDump_Harmonix_RB3_Keyboard_for_Nintendo_Wii.txt](https://github.com/Santroller/Santroller/files/14610420/DescriptorDump_Harmonix_RB3_Keyboard_for_Nintendo_Wii.txt)