Santroller/Santroller #147: [Issue]: Saving settings/writing to device seems to be broken on Linux.

luluco250, 2024-03-09:
### Please describe your issue.

I was able to perform an initial flashing over an Ardwiino kit (only a main Pico board, no peripheral board) but found that the configurator would never actually finish writing any saved settings.

Subsequent attempts to get it to reflash the Pico proved fruitless as it would never show the option to reconfigure the firmware (even when BOOTSEL was held while connecting, which is what I first did when the configurator complained about "contacting the seller").

I then launched a Windows 10 VM and was able to both reflash the device and save settings to it just fine using USB passthrough, which leads me to believe something may be wrong with the Linux implementation. Maybe something with the udev rules.

I've also noticed the program writes its udev rules to `/usr/lib/udev/rules.d/`, whereas non-package rules should generally be written to `/etc/udev/rules.d/` as to not conflict with anything installed by the system package manager. Requiring the program to run as root is also a bit risky, any bug could cause catastrophic damage to the file system.

It's entirely possible the software failed silently due to some missing package/library/program in PATH from my system, but there's no way to know what exactly it could be. I understand source code for the configurator is not be provided due to limitations concerning commercial use but it would be useful to explain what dependencies it needs (it seems to be a .NET app, even if self-contained that still has some hard dependencies).

### Version

v10.0.166

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-03-09:
If it's not showing up in bootsel that's even more weird since all that looks for is the mounted Pico drive. I do all my testing with Arch Linux, and others have tested it with other distros as well.

All that is really necessary is libusb

luluco250, 2024-03-09:
Interesting, I'm also on Arch and have libusb 1.0.27-1

sanjay900, 2024-03-09:
The udev rules certainly work on arch for me, does the drive show up in your file manager in bootsel mode?

sanjay900, 2024-03-09:
Funny thing is originally it was writing to etc and another Linux user told me to change it lol

luluco250, 2024-03-09:
> The udev rules certainly work on arch for me, does the drive show up in your file manager in bootsel mode?

Yes, the usual uf2 mount shows up.

> Funny thing is originally it was writing to etc and another Linux user told me to change it lol

`/etc` is generally reserved for local administration settings, `/lib` and `/usr/lib` are handled by the package manager.

I'd guess the other Linux user wanted to avoid their personal rules to be polluted by rules of a specific program, but the proper way to do this would be to package the udev rules separately.

sanjay900, 2024-03-09:
Correct, root is only used for copying the udev rule

luluco250, 2024-03-09:
Let me check.

luluco250, 2024-03-09:
Gets stuck like this (just tried changing led brightness, but it happens with any change):

[image]

luluco250, 2024-03-09:
Interestingly I *can* set the brightness in the preview, it's when saving settings that it bugs out. It seems to just turn off my device and then does nothing.

After reconnecting my settings are reset to the last time I saved them (in this case in Windows) and `Save Settings` is permanently greyed out.

luluco250, 2024-03-09:
Oh I see what's happening, it saved the settings once I manually went into KDE's mount tray icon and clicked "mount and open". Maybe you need to mount the device manually if it doesn't get mounted by the system automatically?

sanjay900, 2024-03-09:
Yeah, right, that's exactly what the problem is, most distros have auto mounting but I suspect yours doesn't

luluco250, 2024-03-09:
KDE just doesn't mount by default, I think Gnome doesn't either.

Most distros rely on udiskie to handle automounts, but it doesn't come by default on Arch.

sanjay900, 2024-03-10:
I have added a thing to the ui to point out it needs to be mounted in 10.0.167

closed 2024-03-10 by sanjay900