Santroller/Santroller #148: Drum / Cymbal pads (Analog) (Other microcontrollers)

goku1205, 2024-03-10:
### Please describe your issue.

Hi, I know there is a problem with using the multiplexer in the drum kit. I tried to do it with the new method you posted on the website, which involves soldering a 1-ohm resistor to the piezo and connecting it to the digital link in the Pico. Unfortunately, after hitting the button, there is no response. Maybe I'm doing something wrong or not setting it up correctly?

### Version

v10.0.163

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-03-10:
I never actually tried that method myself, i was just told by someone else that it works.
Interestingly, after playing around it seems my multiplexer code does actually work, so you can try the multiplexer option if you want and see if that works for you?

goku1205, 2024-03-12:
The method with connecting the piezo to the digital input doesn't work anymore, I've checked it before. I tested it with the multiplexer the day before yesterday, and unfortunately, it didn't work."

sanjay900, 2024-03-12:
What did the multiplexer do?

lordabarth, 2024-03-27:
Hi, what is the best microcontroller setup for the drums to have 7 analog inputs?I have tried the peripheral setup with 2 RPI pico but is only adding digital inputs. Is it possible to add support for Bluepill STM32F103C8T6 with 10 analog inputs?

sanjay900, 2024-03-27:
use an arduino pro micro

goku1205, 2024-03-28:
When connecting 2 piezo, it's okay. It shows values ​​and everything can be set. When I connect all 6 Piezos, none of them show values. It's as if they're all working as one again.

sanjay900, 2024-03-28:
Yeah I'll have to mod a drum kit then, I have no idea what's going on

goku1205, 2024-04-25:
Hi, I found a Pro Micro clone at 
https://pl.aliexpress.com/item/1005006426010899.html?spm=a2g0o.productlist.main.3.20233b93Hi0L6v&algo_pvid=a70d56af-2d68-487a-b15e-d48cc81a7123&utparam-url=scene%3Asearch%7Cquery_from%3A&gatewayAdapt=glo2pol

Will this board be compatible?

sanjay900, 2024-05-04:
If you want to get the multiplexer working, check the updated drum guide

goku1205, 2024-05-06:
At the moment, I'm working on a project for a Pro Micro controller. It seems that a 1M resistor might be too large for some piezos. Today, I connected them in the same configuration as with the Pico, and encountered the same issue. Now, I've connected two piezos directly without resistors, and everything seems to be working without any noise for now. In the evening, if I manage to, I'll reconnect the old configuration, excluding the three resistors connected to the previously malfunctioning piezos.

closed 2024-05-24 by sanjay900