Santroller/Santroller #149: [Issue]: fails every time i try flash

NikonUK, 2024-03-11:
### Please describe your issue.

if i try to flash the pi pico no matter what it fails. weather its on an earlier version of santroller or latest
[image]
also the "raise issue on github" button redirects to a dead page

### Version

all versions (including latest)

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

NikonUK, 2024-03-11:
my current fix is to just use GuitarConfigurator until my pc wants to work with it. it works on my laptop (both linux and windows) so idk

sanjay900, 2024-03-11:
delete everything from `%appdata%/SantrollerConfigurator` and try again

MorleyGarcia, 2024-03-12:
I'm having this same issue as well.

sanjay900, 2024-03-12:
This happening on the initial config, or once you have made changes?

sanjay900, 2024-03-12:
And when you dismiss this dialog, what does the progress bar say?

sanjay900, 2024-03-12:
When v10.0.171 comes out, could you please get me the text in that box? hopefully it won't be empty.

MorleyGarcia, 2024-03-12:
> This happening on the initial config, or once you have made changes?

For me its happening on the initial config, and when I dismiss the dialog the progress bar says that its searching for the device

sanjay900, 2024-03-12:
@MorleyGarcia v10.0.171 is out now, can you get me the error from  that please?

MorleyGarcia, 2024-03-12:
This is what I see for the initial config
[image]

And then this is what I see when I dismiss the dialog
[image]

sanjay900, 2024-03-12:
How much space is free on your C drive at the moment?

sanjay900, 2024-03-12:
and whats in the `%appdata%/SantrollerConfigurator/Santroller/.pio/build` directory

MorleyGarcia, 2024-03-12:
> How much space is free on your C drive at the moment?

I have about 146GB free

> and whats in the `%appdata%/SantrollerConfigurator/Santroller/.pio/build` directory

I don't see a .pio folder in `%appdata%/SantrollerConfigurator/Santroller/ but there is a pio_packages folder with the below contents.

[image]

sanjay900, 2024-03-12:
open a command prompt from `%appdata%/SantrollerConfigurator/Santroller`
and run `../python/python.exe -m platformio run --target upload --environment pico --upload-port D:/` replacing `D:/` with the drive letter from your pico when its in boot mode

MorleyGarcia, 2024-03-12:
I figured out my issue, so I was using RetroCult Mod's board and I didn't realize he recently updated his firmware for his boards so I had to use his program. 

@NikonUK If by any chance you are using RetroCult's board go to his website and download his program.

sanjay900, 2024-03-12:
That works since it totally sidesteps the broken bits lol

sanjay900, 2024-03-12:
Do your PCs potentially have firewalls that are blocking network connections?

NikonUK, 2024-03-13:
> I figured out my issue, so I was using RetroCult Mod's board and I didn't realize he recently updated his firmware for his boards so I had to use his program.
> 
> @NikonUK If by any chance you are using RetroCult's board go to his website and download his program.

nope. using a raspberry pi pico

also yeah my pcs c drive was full. it works now

sanjay900, 2024-03-13:
It should have warned you that you don't have enough space, ill have to investigate that.

closed 2024-03-13 by sanjay900