Santroller/Santroller #153: [Issue]: PS2 World Tour Guitar problem

RaoulBe, 2024-03-20:
### Please describe your issue.

I am trying to build a PS2 adapter according to the instructions (https://santroller.tangentmc.net/wiring_guides/ps2.html). Hopefully I wired it all right, and I succeeded to program the Pico to be a PS2 adapter.
I tried it out with a couple of Guitar Hero controllers for PS2 (World Tour and Kramer Striker). Both of them light up and the led is blinking, but very faintly, and it doesn't connect to the guitar (wirelessly). Are the wireless PS2 guitars supported, or do I need to build something else? I noticed that the Rumble pin is not used.

### Version

10.0.183

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

PS2

sanjay900, 2024-03-21:
Wonder if it isnt providing enough power for the wireless adapter
can i see the bottom of the adapter?

RaoulBe, 2024-03-21:
Hello and thanks for your response

Here are the dongles and my build.
I measured the voltage on pin 5, and get 3.27, so that should be close enough to 3.3.
I just received some PCBs, and I will try to solder the components, but the circuit should be the same.

/Raoul

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

can i see the bottom of the dongle?

RaoulBe, 2024-03-21:
[image]
[image]
[image]

sanjay900, 2024-03-21:
https://github.com/Santroller/Santroller/issues/104
others have had success with ps2 adapters, so im not actually sure whats going on

RaoulBe, 2024-03-21:
Hi again 

Strange. I built a new one on a PCB, wired in the same way, and same problem persists. 
Would it help if I ask in the Discord instead? Surely someone has encountered the same problem, when following the guide.
My config: 
SCK->GP6, MOSI->GP3, MISO->GP4, ACK->GP7+3.3V on Pico over 1kOhm resistor,ATT->GP10+ 3.3V on Pico over 1kOhm resistor, 3.3V -> 3.3V OUT and GND->GND)
Santroller configured the Pico as PS2, Guitar Controller, and mapped the pins as above.

/Raoul

sanjay900, 2024-03-24:
Resolved by hooking up to VBUS instead of 3.3v, seems these need more juice.

closed 2024-03-24 by sanjay900