Santroller/Santroller #154: [Issue]: MPR121 not detected

GrandpaVape, 2024-03-21:
### Please describe your issue.

I'm using an official pi pico with one of [these](https://a.co/d/0dMlv03) breakout boards (similar to the discontinued SparkFun variant). Using [Adafruit's CircuitPython example](https://learn.adafruit.com/adafruit-mpr121-12-key-capacitive-touch-sensor-breakout-tutorial/python-circuitpython), I can see the board on address 0x5A responding to touches on all pins.

However, when I use SantrollerConfigurator using the same SDA/SCL pins (GP18/GP19), it doesn't show as detected. I've tried 2 of them already and they work in CircuitPython, but not in Santroller. Any ideas?

### Version

v10.0.186

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-03-21:
Odd, I know my mpr121 breakouts work so I wonder what's different with yours

GrandpaVape, 2024-03-21:
@sanjay900 Which ones are you using?

sanjay900, 2024-03-21:
can you throw me the contents of the file `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`

GrandpaVape, 2024-03-21:
```
#define CONFIGURATION {0xB,0x1C,0x80,0x10,0x1,0x20,0x1,0x50,0x1,0x58,0x1,0x60,0x5,0x70,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x1,0xC0,0x1,0xA,0xE8,0x1,0xA,0x8A,0x2,0x0,0x92,0x2,0x0,0xB0,0x2,0x1,0xF8,0x2,0x1,0x80,0x3,0x1F,0x90,0x3,0x1,0x98,0x3,0x5,0xA0,0x3,0x1,0xA8,0x3,0x12,0xB0,0x3,0x13,0x3}
#define CONFIGURATION_LEN 61
#define SWAP_SWITCH_FACE_BUTTONS true
#define WINDOWS_USES_XINPUT true
#define RPCS3_COMPAT false
#define XINPUT_AUTH false
#define INPUT_QUEUE false
#define POLL_RATE 0
#define WT_SENSITIVITY 5
#define INPUT_DJ_TURNTABLE_POLL_RATE 10000
#define INPUT_DJ_TURNTABLE_SMOOTHING false
#define LED_BRIGHTNESS 31
#define ABSOLUTE_MOUSE_COORDS false
#define ARDWIINO_BOARD "pico"
#define EMULATION_TYPE 0
#define DEVICE_TYPE 1
#define USB_HOST_STACK false
#define USB_HOST_DP_PIN -1
#define DIGITAL_COUNT 0
#define LED_COUNT 0
#define LED_COUNT_PERIPHERAL 0
#define LED_COUNT_STP 0
#define LED_COUNT_PERIPHERAL_STP 0
#define ADC_PINS {}
#define ADC_COUNT 0
#define TICK_SHARED \
    
#define TICK_DETECTION \
    
#define TICK_RESET \
    
#define TICK_PS3 \
    
#define TICK_PS3_WITHOUT_CAPTURE \
    
#define TICK_PC \
    
#define TICK_PS4 \
    
#define TICK_XINPUT \
    
#define TICK_XBOX_ONE \
    
#define HANDLE_AUTH_LED \
    
#define HANDLE_PLAYER_LED \
    
#define HANDLE_LIGHTBAR_LED \
    
#define HANDLE_RUMBLE \
    
#define HANDLE_RUMBLE_EXPANDED \
    
#define HANDLE_KEYBOARD_LED \
    
#define PIN_INIT_PERIPHERAL \
    
#define PIN_INIT \
    
#define LED_INIT \
    \
	\
	
#define LED_COUNT_MPR121 0

#define TWI_1_SDA 18
#define TWI_1_SCL 19
#define TWI_1_CLOCK 400000

#define MPR121_TWI_PORT TWI_1
#define MPR121_TOUCHPADS 5
#define MPR121_DDR 0
#define MPR121_ENABLE 0
```

Hmm, MPR121_ENABLE 0 doesn't look correct?

sanjay900, 2024-03-21:
No thats perfectly fine, thats only for if you are using GPIO inputs

sanjay900, 2024-03-21:
[firmware.uf2.zip](https://github.com/Santroller/Santroller/files/14712797/firmware.uf2.zip)
try this

GrandpaVape, 2024-03-21:
[image]
Still not detected

sanjay900, 2024-03-22:
[firmware.uf2.zip](https://github.com/Santroller/Santroller/files/14715187/firmware.uf2.zip)

sanjay900, 2024-03-22:
Actually i just looked at your circuit python lib and its literally the code i based my code on haha

sanjay900, 2024-03-22:
[firmware.uf2.zip](https://github.com/Santroller/Santroller/files/14715211/firmware.uf2.zip)
i noted they have a delay after the soft reset so this has that in it

sanjay900, 2024-03-22:
[firmware.uf2.zip](https://github.com/Santroller/Santroller/files/14715252/firmware.uf2.zip)
i dropped the i2c frequency

sanjay900, 2024-03-22:
My breakout looks pretty much identical to yours FYI

GrandpaVape, 2024-03-22:
yeah still nothing

sanjay900, 2024-03-22:
you got a second pi pico?

GrandpaVape, 2024-03-22:
I have one I need to solder some pins to real quick, or I have an [RP2040-Zero](https://a.co/d/exz8G4v) that's ready to go if you can change the pins to one of the available ones. Otherwise gimme like 10 mins to whip out my Pinecil.

FWIW, I first tried this using a [PhunkyCustom's PCB](https://www.etsy.com/listing/1481506753/monoboard-rp-rp2040-integrated-pcb-for) and it wasn't working, but that was before your changes.

sanjay900, 2024-03-22:
We would be setting up a main pico with the mpr attached
and a second probe pico
probe GP4 -> main GP0
probe GP5 -> main GP1
probe GND -> main GND

GrandpaVape, 2024-03-22:
oh yeah, I can use the rp2040zero as the probe connected to the Pi Pico. Lemme wire that up rq.

sanjay900, 2024-03-22:
https://github.com/raspberrypi/debugprobe/releases/download/debugprobe-v2.0/debugprobe_on_pico.uf2
throw this on the probe
[firmware.uf2.zip](https://github.com/Santroller/Santroller/files/14715337/firmware.uf2.zip)
and this on the main

sanjay900, 2024-03-22:
then you should be able to open a serial terminal using something like putty, point it at the probe and see logs from main

GrandpaVape, 2024-03-22:
[image]
It looks like I'm getting garbage. Are my settings wrong or does that mean my wiring is bad?

GrandpaVape, 2024-03-22:
115200, just like u said

sanjay900, 2024-03-22:
odd, it shouldn't matter what order you plug it in either.

GrandpaVape, 2024-03-22:
Wait im dumb, lemme check device manager

sanjay900, 2024-03-22:
Oops, i got the wiring wrong
probe GP4 -> main GP1
probe GP5 -> main GP0

I always forget to flip tx and rx

GrandpaVape, 2024-03-22:
Ok, so now I get "Console type: 0 | Console type: 9" every time I plug in the main
[image]

sanjay900, 2024-03-22:
Oh i think i see the problem
can you actually map an input on your mpr121 to something?

GrandpaVape, 2024-03-22:
Oh my god was that the problem in the first place :0
[image]

sanjay900, 2024-03-22:
good stuff haha, ill get that fixed

GrandpaVape, 2024-03-22:
I had tried binding it to the slider several times before I opened this issue, why didn't that fix it?

sanjay900, 2024-03-22:
I have no idea, but ill fix it now so that it just works even without it being mapped to anything

sanjay900, 2024-03-22:
v10.0.188 should have this fixed

GrandpaVape, 2024-03-22:
You're a king and a legend. Should I close this, or wait to confirm it's fixed with the new version?

sanjay900, 2024-03-22:
Ill close it now

closed 2024-03-22 by sanjay900