Santroller/Santroller #156: [Issue]: Recieved error when attempting to install firmware on Pi Pico

ahSamyono, 2024-03-26:
### Please describe your issue.

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/60d6ae81fcc73c34b1493ca9e261695e471bc0c2.zip@1.11.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Unpacking
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_sha.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\usbdsec.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\xsm3.c.o
Generating linkerscript C:\Users\Ahmad Samyono\AppData\Roaming\SantrollerConfigurator\Santroller\.pio\build\pico/memmap_default.ld
Compiling .pio\build\pico\lib9e3\SPI\SPI.cpp.o
Compiling .pio\build\pico\lib071\USB-Host-Shield-20\AMBX.cpp.o
src\shared\main\shared_main.cpp: In function 'uint8_t tick_inputs(void*, USB_LastReport_Data_t*, uint8_t)':
src\shared\main\shared_main.cpp:1539:9: error: 'TICK_OG_XBOX' was not declared in this scope
 1539 |         TICK_OG_XBOX;
      |         ^~~~~~~~~~~~
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1

### Version

10.0.201

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-03-27:
Oops, I must have missed a line somewhere I'll cut a fix in a moment

sanjay900, 2024-03-27:
Should be fixed when v10.0.202 comes out.

closed 2024-04-01 by sanjay900