Santroller/Santroller #157: [Request]: Enable face (and shoulder) button layout swap for Classic Controller (Pro)

Soundtoxin, 2024-03-30:
### Please describe your request.

I have this [retromodding guitar adapter](https://shop.retrocultmods.com/products/v2-wii-adapter-for-clone-hero-and-rb4-rock-band-4) for Wii guitars, I've reprogrammed it to be a standard gamepad in the configurator and then plugged my Classic Controller Pro into it. Most stuff just works by the looks of it, however I noticed that with gamepad-tester, my Nintendo b buttons is the Xbox b button, in other words it doesn't match the physical location. My south face button shows up as the east face button. While some prefer this, it makes switching between controllers on PC more confusing, and most third party controllers with an XInput mode will use the physical Xbox layout even if they have the Nintendo layout labels on the buttons. So, an option to swap a/b and x/y around would be great.
I also noticed that in gamepad-tester, what I would call the bumpers show up as the triggers. This might make default binds confusing in some games, so an option to swap them would be ideal. On the non-Pro Classic Controller this makes perfect sense since it's L and R that have analog function, not ZL and ZR, but on the Pro it's all digital anyway, so similar to the face buttons, I'd prefer the physical layout to be more Xbox-like for more typical PC controls, kinda like using a Switch Pro controller on PC (plus SteamOS and probably Big Picture Mode have their own setting to swap those buttons to the Nintendo layout for people who occasionally want that).

If this is already possible, sorry that I missed it. Didn't see it in the GUI. If I ticked the "enable digital output" toggle, I saw I could change the assigned pin, but I wasn't sure what I was getting into there, so I left that alone, and a clean swap back and forth toggle sorta like the current "swap Switch Face buttons" setting would be awesome anyway. Could probably call it "Swap XInput Face Buttons", then maybe put the shoulder button swap in the Wii Extension Inputs section like where mapping joystick to dpad is. Just an idea.

P.S. Sorry I'm stalling on the drum issue, kinda burned out on that project at the moment but I'd still like to get around to it again eventually.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2024-03-30:
There already is an option to swap the face buttons

Soundtoxin, 2024-03-30:
Awesome! I had never used that feature before. I was able to get all the buttons bound exactly how I want now, thank you very much.

If you'd consider preset toggles for these button-swaps like I'd mentioned, I'd like to leave the issue open, but if you consider that out of scope or unnecessary, feel free to close this issue now.

sanjay900, 2024-03-30:
I may look into it but it's probably not worth keeping an issue open for it

closed 2024-03-30 by Soundtoxin