Santroller/Santroller #158: [Issue]: Very weak leds and analog outputs for Tilt axis

ruenes1, 2024-03-31:
### Please describe your issue.

When associating leds or an analog output (e.g., rumble) to the Tilt axis, their intensity is very weak. This does not happen for other analog axis (Whammy), which produces full intensity. The weak output seems to be related to the fact that the Tilt is reported as 'half' axis range, rather than full range (as it is for Whammy) 

### Version

10.0.208

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-03-31:
Not much I can do about that really since I just map the raw value directly to the led

ruenes1, 2024-04-01:
I see now that the problem might be because of using a digital input (mercury switch) as tilt sensor. I understand that, in this case, the raw value is just 0 or 1, thereby producing a very weak output in comparison with the whammy potentiometer, which produces a continous output range. Maybe the digital inputs associated to axis/analog outputs could be scaled/mapped to the output's range, that is from 0..1 -> 0..255?

sanjay900, 2024-04-01:
Realistically, i just need to implement LEDs differently for tilt, since it is very specific in the way it works

sanjay900, 2024-06-15:
I did at one point recently rewrite leds, did that change this?

ruenes1, 2024-06-15:
I updated to the latest version but the problem persists. The issues comes from using a digital input (a mercury switch) for an input that is dealt as analog (tilt). It seems that the firmware just forwards the read value (1, when the mercury switch is activated) to the leds/analog output, which would expect a 255 value. 

Another thing I noticed when upgrading to the latest version (127) is that digital outputs associated to any input (e.g., frets) get 'stuck'. That is, they activate once and never deactivate. This didn't hapen with the version I had before (95). Do you want me to open a separate issue for this?

sanjay900, 2024-06-15:
Can just keep managing that here, ill have a look in a moment

sanjay900, 2024-06-16:
Can you see if v10.1.129 makes a difference once it finishes building?

ruenes1, 2024-06-16:
Thanks. I've updated to ver. 130 and now the digital tilt is properly scaled to the analog output. 

However, digital outputs for button inputs still get stuck. For example, if I associate a digital output (e.g., a plain non-addressable led) to strum up button, it gets activated the first time I push the strum up, but it does not deactivate afterwards and it stays on forever.

sanjay900, 2024-06-16:
Can you send me a copy of %appdata%/SantrollerConfigurator/santroller/include/config_data.h

It'll only exist just after saving settings

sanjay900, 2024-06-16:
try v10.1.131

ruenes1, 2024-06-16:
I've just updated to the latest version (132) and now it works! Thanks!

sanjay900, 2024-06-17:
is the original issue also fixed now as well?

ruenes1, 2024-06-17:
Yes! Feel free to close the issue

closed 2024-06-17 by sanjay900