Santroller/Santroller #159: [Issue]: Wii GH Drums Don't Work Well on GH Games on Xbox 360

cahamilton409, 2024-03-31:
### Please describe your issue.

I recently got some GH5 wii drums and have them hooked up to a Santroller Wii adapter running the latest firmware. They work great on both Clone Hero and specifically Rock Band games on my Xbox 360, so I don't think there's any hardware issues on my end. The drums don't work very well in any of the Guitar Hero games that I have tried though (I tried World Tour and Warriors of Rock). In the Guitar Hero games, it only picks up about 2/3 of the notes that I play.

If you get around to looking at this, I'd be happy to help test out anything. I have a both a modded Xbox 360 and modded PS3, and I wouldn't mind testing out any potential fixes on both platforms! 

Thanks! Love the tool!

### Version

v10.0.208

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2024-04-01:
You probably need to adjust your debounce, for drums you actually want a higher debounce, as you need to stretch out the time the drum reports itself as being hit for

cahamilton409, 2024-04-01:
Tried putting the debounce up to 20ms, and it works great now! Thanks for the quick reply!

closed 2024-04-01 by cahamilton409