Santroller/Santroller #161: [Issue]: RGH Xbox 360 Crashes with Multiple Santrollers

cahamilton409, 2024-04-06:
### Please describe your issue.

I have a couple of guitars running the Santroller firmware, which have been working well individually on Xbox 360. It seems like when I try to use multiple Santrollers at once though, the console will often crash pretty soon after plugging them in (There have been a couple times where I have been able to play for an extended period of time with multiple Santrollers plugged in, but it has been rare). I have not really had any stability issues with using a single Santroller in conjunction with wireless xbox 360 guitars, which makes me think that it's something weird with using multiple Santrollers at once maybe. I also have a PS3 where I have never had any stability issues using two Santrollers. I would appreciate any suggestions or insights.

Thanks!

### Version

v10.0.208

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-04-07:
Not really a lot I can do unfortunately, what game?

cahamilton409, 2024-04-07:
No worries. It doesn't really seem to be specific to any game. I've had it freeze on the Aurora menu many times and in a variety of Guitar Hero and Rock Band games. I may eventually try hooking up authentication to the controllers and try on an unmodded Xbox 360.

sanjay900, 2024-04-24:
are you using the patch via jrunner?

cahamilton409, 2024-04-25:
I bought the console second-hand, so I'm not exactly sure what was done to hack the console. I know that it is RGH 3 via Jrunner. What do you mean by "patch"? I have confirmed that it doesn't have the USB XL patch if that is what you are referring to.

sanjay900, 2024-04-25:
There's two ways to use usbdsecpatch. You can either enable it in jrunner or you can just put the xex in dash launch, so was curious which method you used

cahamilton409, 2024-04-25:
Oh gotcha, I just put the xex in DashLaunch.

sanjay900, 2025-05-08:
This should be a bit better now, just make sure for devices 3 and 4 you wait for the previous controller to show up as a player   before you plug the next controller in

closed 2025-05-08 by sanjay900