Santroller/Santroller #162: [Issue]: Error writing Fortnite Festival guitar

CanOfPepsii, 2024-04-07:
### Please describe your issue.

Log Details:

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/60d6ae81fcc73c34b1493ca9e261695e471bc0c2.zip@1.11.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Unpacking
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
In file included from src\shared\main\max1704x.cpp:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\pico\bt.cpp:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\pico\bt_rx.cpp:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\shared\main\mpr121.cpp:3:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from include/hid.h:4,
                 from src\shared\hid\hid.cpp:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
*** [.pio\build\pico\src\pico\bt.cpp.o] Error 1
*** [.pio\build\pico\src\pico\bt_rx.cpp.o] Error 1
In file included from src\pico\xinput_device.cpp:10:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\shared\config\config.c:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from include/controllers.h:1,
                 from src\shared\controllers\controllers.c:1:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
*** [.pio\build\pico\src\shared\controllers\controllers.c.o] Error 1
*** [.pio\build\pico\src\shared\main\mpr121.cpp.o] Error 1
*** [.pio\build\pico\src\shared\config\config.c.o] Error 1
In file included from src\shared\main\adxl345.cpp:4:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
*** [.pio\build\pico\src\shared\main\adxl345.cpp.o] Error 1
In file included from src\pico\io.cpp:9:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\pico\pins.cpp:6:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from include/reports/og_xbox_reports.h:6,
                 from include/xbox.h:4,
                 from include/descriptors.h:9,
                 from src\pico\xinput_host.cpp:32:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
In file included from src\pico\main.cpp:12:
include/config.h:315:2: error: #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
  315 | #error missing HID_AXIS_COUNT and HID_BUTTON_COUNT
      |  ^~~~~
*** [.pio\build\pico\src\pico\pins.cpp.o] Error 1
*** [.pio\build\pico\src\shared\main\max1704x.cpp.o] Error 1
*** [.pio\build\pico\src\shared\hid\hid.cpp.o] Error 1
*** [.pio\build\pico\src\pico\io.cpp.o] Error 1
*** [.pio\build\pico\src\pico\xinput_device.cpp.o] Error 1
*** [.pio\build\pico\src\pico\xinput_host.cpp.o] Error 1
*** [.pio\build\pico\src\pico\main.cpp.o] Error 1
========================== [FAILED] Took 7.11 seconds ==========================

### Version

v10.0.215 and v10.0.213

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-04-07:
Weird I was sure I had fixed this, I'll look later

sanjay900, 2024-04-08:
Fixed when v10.0.216 comes out, should be out in 15m or so

closed 2024-04-08 by sanjay900