Santroller/Santroller #164: [Issue]: Tilt does not work in wireless mode

XxyZwW, 2024-04-21:
### Please describe your issue.

I made a wireless adapter with 2 RPpW, I put an adxl345 in the transmitter and it does not activate the overdrive in rock band 4, I thought it could be an error with the adxl345, I changed it and installed 2 Mercury Ball tilt, low configure in GPIO 16 ( the other pole to gnd) I did the test again and it doesn't activate the overdrive in RB4 either, I tried testing wired and it does activate the overdrive, so I deduce that the problem is only in wireless mode.

### Version

10.0.223

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Wii

sanjay900, 2024-04-21:
interesting i thought i fixed this problem recently but i guess not

sanjay900, 2024-04-22:
What console are you playing on?

XxyZwW, 2024-04-22:
On PS4, I made a wireless adapter for Wii guitars

sanjay900, 2024-04-22:
have you got it in RB or gh mode

XxyZwW, 2024-04-22:
Estoy emulando guitarra de Guitar Hero

sanjay900, 2024-04-24:
Can you try the latest release? this might be fixed now

XxyZwW, 2024-04-26:
Sorry, I hadn't had time to test, but yes, I confirm that yes, the Tilt now works in wireless mode, thank you very much for your work brother.

closed 2024-04-26 by sanjay900