Santroller/Santroller #165: [Request]: Option to force authentication on PS4/5

UVClay, 2024-04-24:
### Please describe your request.

I haven’t tested this yet so my apologies if this is already baked in.  With FF requiring either a Riffmaster, RB4 Jaguar, or RB4 Strat, is there an option to force the firmware to attempt authentication?  I’ve been able to get my guitar working in PS3 mode using rb4.app, but having two extra Picos in the mix makes things real scuffed.  I’m getting my Mayflash dongle later today so I wanted to ask up front before I added USB input to my Pico.

### What operating system are you using?

Linux

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-04-24:
That's not the problem. I need to emulate PS4 guitars and I currently don't do that because until now there hasn't been a reason to do so. I'll do that soon hopefully but it'll take time

closed 2024-04-24 by sanjay900

sanjay900, 2024-04-25:
I have got my hands on a PS4 so ill probably find a way to get to this today.