Santroller/Santroller #167: [Issue]: When on Mac, can't revert back to Arduino

iamnoshade, 2024-04-30:
### Please describe your issue.

Im having an issue where I set my guitar controller to emulate a PS3 controller but now it's stuck in that mode and when I press "revert back to Arduino", the loading bar is stuck at 10% and never moves.

### Version

10.1.32

### What operating system are you using?

Mac OS

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-04-30:
since its a pi pico you can just unplug the pico, hold the boot button and plug it back in.
I don't know what you mean by set it to emulate a ps3 controller, there is no explicit setting for that in santroller.

sanjay900, 2024-04-30:
Ah yeah, that makes sense it won't respond to some commands in that mode

closed 2024-04-30 by sanjay900