Santroller/Santroller #170: [Issue]: USB Adapter maps triggers to the wrong axis when plugged into a PS3

pajaritia, 2024-05-09:
### Please describe your issue.

Using Santroller Configurator v10.1.48
I configures a RP Pico as USB Adapter, plugging a Xbox360 Racing Wheel via USB the inputs look correct on the Santroller Configurator and in windows.
When connected to a PS3 the triggers (L2 R2) are mapped to the Circle and Triangle buttons pressure sensitive axis.
All the other inputs are correctly mapped and work as expected so I think the adapter is working okay.
[image]
[image]

### Version

v10.1.48

### What operating system are you using?

Windows 10

### os-other

PS3

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-05-09:
Must be something wheels do differently I guess

pajaritia, 2024-05-09:
I tested with a xbox360 compatible regular gamepad and it's the same issue, triggers are mapped to the pressure of Triangle and Circle.

sanjay900, 2024-05-09:
can you send me a copy of `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`
it will only exist after you hit save settings

pajaritia, 2024-05-09:
[config_data.zip](https://github.com/Santroller/Santroller/files/15257342/config_data.zip)
Here is the file requested, thanks for looking into it.

sanjay900, 2024-05-09:
Should be fixed with v10.1.49 comes out

closed 2024-05-09 by sanjay900